UniversalRobots/Universal_Robots_ROS2_Driver #957: unabele to execute path useing moveit on real hardwear

bodkal, 2024-04-02:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

ros2 run ur_client_library start_ursim.sh -m ur10e

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
whan i trray to execute path using moveit with real hardwer i get:  MoveGroupInterface::execute() failed or timeout reached.

## Issue details
i can move the arm useng the topic /urscript_interface/script_command. in addisen if i move the arm in the docker i can see it moveing in rviz. on fake harwer it working as expected.

logs for ur_robot_driver:

```
[INFO] [launch]: All log files can be found below /home/picko/.ros/log/2024-04-02-18-17-51-138774-picko-8411
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [8416]
[INFO] [controller_stopper_node-3]: process started with pid [8418]
[INFO] [ur_ros2_control_node-1]: process started with pid [8414]
[INFO] [urscript_interface-4]: process started with pid [8420]
[INFO] [robot_state_publisher-5]: process started with pid [8422]
[INFO] [spawner-6]: process started with pid [8424]
[INFO] [spawner-7]: process started with pid [8426]
[INFO] [spawner-8]: process started with pid [8428]
[INFO] [spawner-9]: process started with pid [8430]
[INFO] [spawner-10]: process started with pid [8432]
[INFO] [spawner-11]: process started with pid [8434]
[dashboard_client-2] [INFO] [1712071071.591068025] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ur_ros2_control_node-1] [WARN] [1712071071.592408810] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[controller_stopper_node-3] [INFO] [1712071071.592509710] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1712071071.592701158] [resource_manager]: Loading hardware 'ur10e' 
[robot_state_publisher-5] [INFO] [1712071071.593090806] [robot_state_publisher]: got segment base
[robot_state_publisher-5] [INFO] [1712071071.593154898] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1712071071.593159571] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-5] [INFO] [1712071071.593162094] [robot_state_publisher]: got segment flange
[robot_state_publisher-5] [INFO] [1712071071.593164469] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-5] [INFO] [1712071071.593166862] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-5] [INFO] [1712071071.593169148] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-5] [INFO] [1712071071.593171367] [robot_state_publisher]: got segment tool0
[robot_state_publisher-5] [INFO] [1712071071.593173565] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-5] [INFO] [1712071071.593175750] [robot_state_publisher]: got segment world
[robot_state_publisher-5] [INFO] [1712071071.593177865] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-5] [INFO] [1712071071.593180080] [robot_state_publisher]: got segment wrist_2_link
[... 208 lines cut ...]
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] active    
io_and_status_controller[ur_controllers/GPIOController] active   
```

## Steps to Reproduce
i run in 3 terminals:

ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.56.101 launch_rviz:=false use_fake_hardware:=false 

ros2 run ur_client_library start_ursim.sh -m ur10e

 ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10e launch_rviz:=true

then try to move the arm useing the rviz plagin (move from code dosent work as well)

## Expected Behavior
arm move to goal.

## Actual Behavior
arm dosent move.

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-04-08:
Did you start the external control program on the teach pendant after starting the driver (e.g. after `ros2 launch ur_robot_driver ur_control.launch.py [...]`)? When this is started you should see the lines
```
[ur_ros2_control_node-1] [INFO] [1712574636.743142775] [UR_Client_Library:]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1712574636.743210885] [UR_Client_Library:]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1712574636.789046949] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
```
in your controller log. Alternatively you can use [headless mode](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/ROS_INTERFACE.html#headless-mode).

bodkal, 2024-04-08:
tanks for the replay.
I'm using the script from the ur_client_library, so the external control should be enabled from the beginning.
[image]

And it looks like the external control is enabled.
[image]

i dont see the log you provided in my terminal. any idea?

RobertWilbrandt, 2024-04-10:
URsim has the urcap installed, but no program using it and also doesn't start it directly (which wouldn't work, because you have to start the driver on the PC first). Please look at the already linked [headless mode](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/ROS_INTERFACE.html#headless-mode) if you want to start the program automatically on driver startup.

fmauch, 2024-07-10:
@bodkal Since the above comment seems to be an adequate solution, I will go ahead and close this issue. If the issue still persists and/or you would like to comment something more, feel free to do so and/or reopen this issue.

closed 2024-07-10 by fmauch