UniversalRobots/Universal_Robots_ROS2_Driver #965: The test_joint_trajectory_controller not work with ur_robot_driver for Humble but work with Foxy

nqduy35, 2024-04-11:
### Affected ROS2 Driver version(s)

Humble branch 

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot, URSim in docker

### Robot SW / URSim version(s)

5.14

### How is the ROS driver used.

Others

### Issue details

# Summary + Issue + Steps
Trying installing ur_ros_driver on Docker ubuntu 22.04 with ros 2 Humble, all installation steps are succeeded and the ur_ros_driver runs without any errors. However, when I test with test_joint_trajectory_controller, the robot does not move. Note that my host computer runs ubuntu 20.04 with ros 2 Foxy, so I can produce several tests as follows:  

1. Foxy ur_ros_driver + Foxy test_joint_trajectory_controller + UR3e  => Success
2. Foxy ur_ros_driver + Humble test_joint_trajectory_controller + UR3e => Success
3. Foxy ur_ros_driver + Foxy test_joint_trajectory_controller + URsim  => Success
4. Foxy ur_ros_driver + Humble test_joint_trajectory_controller + URsim  => Success
5. Humble ur_ros_driver + Foxy test_joint_trajectory_controller + UR3e  => Failure
6. Humble ur_ros_driver + Humble test_joint_trajectory_controller + UR3e => Failure
7. Humble ur_ros_driver + Foxy test_joint_trajectory_controller + URsim => Failure
8. Humble ur_ros_driver + Humble test_joint_trajectory_controller + URsim => Failure

=> So i temporarily conclude that the problem is from Humble ur_ros_driver

Example, the following the screen capture of the 8th case. Window 1: the URsim, Window 2: running ur_ros_driver, Window 3: running test_joint_trajectory_controller. As you can see, the UR3e in URsim can detect the ur_ros_driver, but nothing happens when running test_joint_trajectory_controller. 
[image] 

### Accept Public visibility

- [X] I agree to make this context public

MoritzLennartz, 2024-04-12:
i have the same issue, when trying to build the ur driver with humble.
[image]

fmauch, 2024-04-15:
I can't recall how we did things with Foxy, but we use the **scaled** joint trajectory controller as default since quite a while, so you would need to use the `test_scaled_joint_trajectory_controller.launch.py` file.

Please note (If anyone else reads this issue, since @nqduy35 seems to have done that correctly judging from the screenshots) that you will have to move the robot to a valid start pose (upright pose as in the screenshot) and have to make sure that the external_control program is actually running.

@nqduy35 If using the scaled version doesn't work for you, could you please post the output of `ros2 control list_controllers`? It should look as follows (when the external_control program is running)

```
forward_position_controller[position_controllers/JointGroupPositionController] inactive  
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
io_and_status_controller[ur_controllers/GPIOController] active    
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
```

nqduy35, 2024-04-16:
First of all, thank @fmauch for your answer. 

I'm sending you the new screen capture following your instruction. You can see: 
- Left window: UR3e sim is connected but not running.
- Black window: ur_ros2_driver running and showing that the ur3e is ready. 
- Red window: result of the command `$ ros2 control list_controllers` showing that the scaled_joint_trajectory_controller is active. 
- Blue window: result of launching `ur_robot_driver test_scaled_joint_trajectory_controller`

[image]

closed 2024-04-16 by nqduy35

zhichaoliu0921, 2025-03-05:
@fmauch I am having the same issue with Humble + UR10e. I was about to launch the `ros2 launch ur_robot_driver ur_control.launch.py` and see the correct pose in rviz. The output of running `ros2 control list_controllers` is 
```
joint_state_broadcaster            joint_state_broadcaster/JointStateBroadcaster                 active  
joint_trajectory_controller        joint_trajectory_controller/JointTrajectoryController         inactive
forward_velocity_controller        velocity_controllers/JointGroupVelocityController             inactive
io_and_status_controller           ur_controllers/GPIOController                                 active  
forward_position_controller        position_controllers/JointGroupPositionController             inactive
speed_scaling_state_broadcaster    ur_controllers/SpeedScalingStateBroadcaster                   active  
force_mode_controller              ur_controllers/ForceModeController                            inactive
force_torque_sensor_broadcaster    force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster  active  
passthrough_trajectory_controller  ur_controllers/PassthroughTrajectoryController                inactive
freedrive_mode_controller          ur_controllers/FreedriveModeController                        inactive
tcp_pose_broadcaster               pose_broadcaster/PoseBroadcaster                              active  
ur_configuration_controller        ur_controllers/URConfigurationController                      active  
scaled_joint_trajectory_controller ur_controllers/ScaledJointTrajectoryController                inactive

``` 
The PolyScope is in `Remote Control` mode and I was about to ping the PolyScope in my PC. Anyone has any thoughts? Thanks!

urfeex, 2025-03-06:
@zhichaoliu0921 please open a new issue instead of posting in an old, closed issue. In that issue, please post the **exact** commands that you use. Please also see https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/operation_modes.html