UniversalRobots/Universal_Robots_ROS2_Driver #966: arm move to slow in Cartesian movemants

bodkal, 2024-04-14:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

ros2 run ur_client_library start_ursim.sh -m ur10e

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I'm trying to make the arm (UR10e) execute Cartesian paths as fast as possible. However, no matter what I do, the computeCartesianPath function returns a trajectory with joint velocity capped at approximately 0.6 rad/s. I believe that the arm is capable of speeds up to ~3 rad/s per joint.

## Issue details

## Steps to Reproduce

1. Try different start and goal configurations with the same result.
2. Attempt to modify the joint_limits.yaml in ur_description with the same result.
3. Try modifying the ur_servo.yaml in ur_moveit with the same result.

## Expected Behavior
The arm should calculate velocity based on the joint_limits.yaml. am i wrong? diffrent yaml?

## Actual Behavior
Changing the YAML file didn't affect the results.

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-04-15:
I am not familiar enough with MoveIt!-specific configuration to answer this in a full extent, but I can say the following:

* *Velocity* isn't capped, and definitively not to 0.6 rad/s. I ran a couple of tests and reached velocities around 1.4 rad/s during test motions.
* I assume the rather slow motions are due to the conservative acceleration limits used by MoveIt! by default. We increased those for the rolling version in #645 but it might as well affect the Humble version. On Rolling I could achieve peak velocities of above 2 rad/s but at that point the robot slowed down the motion due to its Cartesian safety limits. When performing a motion only in the last joint, I even got around 3.1 rad/s.

Maybe we should backport #645 also to Humble.

fmauch, 2024-04-25:
Closing this since #981 got merged.

closed 2024-04-25 by fmauch