UniversalRobots/Universal_Robots_ROS2_Driver #972: Could we stop the External Control on the UR robot at runtime?

nqduy35, 2024-04-18:
### Affected ROS2 Driver version(s)

All

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.14

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

Using the URCap External Control and ROS 2 ur_ros_driver, I wonder if I can stop the External Control running on my UR robot (
[image]
) at runtime. In more detail: 
- Could I stop or pause the External Control from the ur_ros_driver side?
- Could I stop or pause the External Control from the UR robot side? 

Thank you in advance for your answer. 

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-04-19:
`ros2 service call /io_and_status_controller/hand_back_control std_srvs/srv/Trigger` should do the trick

nqduy35, 2024-04-19:
Nice. Thank you very much @fmauch

closed 2024-04-19 by nqduy35