UniversalRobots/Universal_Robots_ROS2_Driver #974: ProgramState.msg not used in GPIOController

firesurfer, 2024-04-22:
### Affected ROS2 Driver version(s)

All

### Used ROS distribution.

Iron

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

....

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

I just saw that there is a specialized `ur_dashboard_msgs::msg::ProgramState` message. 
It is a bit confusing that the GPIOController uses a `std_msgs::msg::Bool` message instead for the program state

```
program_state_pub_ =
        get_node()->create_publisher<std_msgs::msg::Bool>("~/robot_program_running", program_state_pub_qos);
```

Would be nice if this could be fixed in the long term as for the `robot_mode` and the `safety_mode` the dashboard messages are used. 

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-04-24:
The `robot_program_running` state actually reports whether the robot is connected to the reverse interface. Sorry for the confusing name that goes back quite a wile...

`ur_dashboard_msgs::msg::ProgramState` would report whether **any** program is running on the robot. So, in my opinion the boolean flag does indeed make sense in that context.

firesurfer, 2024-04-24:
I see your point. This makes sense..

closed 2024-04-24 by firesurfer