UniversalRobots/Universal_Robots_ROS2_Driver #975: Planning precise straight line movements for UR10e with ROS2

shrutichakraborty, 2024-04-22:
Hi, I am using the UR10e robot with ROS2 Humble. I have to perform a couple of fine straight line motions for robotic tool changing, where all my waypoints are offsets in X,Y, Z directions. I am able to do this very simply via the teach pendant where I define sppeL movements. However, I need to do this via ROS2 and as there is no cartesian planner, I had been trying the cartesian motion controller proposed by : https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers and that was recommended also in the issues in this repo. 

However, I have found after repeated testing that the IK solver with this repo results in random jumps and vibrations in the robots movement, and doesn't execute clean precise cartesian movement. I would like to know, instead of cartesian commands, if I record the waypoints as joint states instead, and use the `scaled_joint_trajectory_controller`  and the `FollowJointTrajectory` action, will this execute more straight line motions ? I am wondering this as using the joing trajectory controller means I use the robot's internal controller and trajectory generation for the motion and I am assuming is more reliable than other planners. 

If not, what is the best slution? Would using moveit be the only recommended solution for getting clean precise straight line movements? 

Thanks a lot!

firesurfer, 2024-04-22:
You probably want to setup Moveit2 with the Pilzplanner. This planner is also available for ROS2.

fmauch, 2024-06-12:
I'll close this issue as the proposed solution seems to be appropriate for the question.

closed 2024-06-12 by fmauch