UniversalRobots/Universal_Robots_ROS2_Driver #982: Sending urscript in remote control mode while UR_Driver running

Sohaib-Snouber, 2024-04-29:
### Affected ROS2 Driver version(s)

ROS2

### Used ROS distribution.

Iron

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

polyscope

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I have installed the ROS2 ur_driver and the URCaps and both running fine with testing them using the MoveIt, but i have only in rviz just the ur5e without the gripper, so i have read from an issue that it is not easy to configure the gripper in the ur_description and be able to control it, so i have chosen to use the urscript to be able to control the gripper, but till now i have only succeded to control the digital output and be able to move the robot using the urscript by running a python code as the following:

`import socket

def send_urscript(ip, port, script):
    try:
        with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
            s.connect((ip, port))
            s.sendall(script.encode('utf-8'))
            print("Command sent successfully.")
    except Exception as e:
        print(f"An error occurred: {e}")

robot_ip = '10.130.1.100'  # Replace with your robot's IP
port = 30002  # Standard port for URScript commands

urscript_command = '''
def secondaryProgram():
  set_digital_out(1, False)\n
  sleep(5)\n
  set_digital_out(1, True)\n
  sleep(10)\n

end\n
secondaryProgram()
'''

send_urscript(robot_ip, port, urscript_command)

`
after running the ur_driver and running my urcaps external control, i also used the remote control to be able to send urscripts, but when i run the the python code i get what i have written in the code, but after that the urcaps external control will be stopped automatically alone, i have thought of primary and secondary programms and using the information of this site [https://www.universal-robots.com/articles/ur/interface-communication/remote-control-via-tcpip/] i did not get any other response using port like 30001, 30002 or 30003, the other ports did not work for remote control (I could not receive the true digital output by using 30004, 30011, etc). 

Another problem is that i tried in this way to control the gripper (2F Adaptive gripper) that is connected to the wrist of the roboter with these type of commands e.x. "rq_open()"  in urscript, but it didn't work (I have only activated the gripper from the teach pendant after i turned on the device).

One more thing that is maybe be important to say is that while running the ur_driver a two repeated message is shown that about 0.5 - 1 ms the connection to reverse interface dropped as shown 

> [ur_ros2_control_node-1] [INFO] [1714412116.518364831] [UR_Client_Library:]: Connection to reverse interface dropped.
[ur_ros2_control_node-1] [INFO] [1714412116.518765602] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
[ur_ros2_control_node-1] [INFO] [1714412117.378185573] [UR_Client_Library:]: Connection to reverse interface dropped.
[ur_ros2_control_node-1] [INFO] [1714412117.378787896] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.

Finally when the python code is executed then the "connection to reverse interface dropped" message is only received.

Thanks for all your help and support.

### Accept Public visibility

- [X] I agree to make this context public

Sohaib-Snouber, 2024-04-30:
Update from working on this problem:

I have used what here in this link given: [https://www.universal-robots.com/articles/ur/programming/secondary-program/](url)
I mean i have changed my urscript, instead of using def, I have used "sec" as a definition of a secondary program. After running the ur_driver and  running the urcaps external control and using remote control, then the python code has been executed, then the ur_driver has failed(and trying to reconnect) and i turn it off and run it again, i have got this as output log: 

> sohaib@sohaib-GE75-Raider-10SF:$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=10.130.1.100 launch_rviz:=false
[INFO] [launch]: All log files can be found below /home/sohaib/.ros/log/2024-04-30-12-22-10-394976-sohaib-GE75-Raider-10SF-10503
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [10510]
[INFO] [controller_stopper_node-3]: process started with pid [10512]
[INFO] [ur_ros2_control_node-1]: process started with pid [10508]
[INFO] [urscript_interface-4]: process started with pid [10514]
[INFO] [robot_state_publisher-5]: process started with pid [10516]
[INFO] [spawner-6]: process started with pid [10518]
[INFO] [spawner-7]: process started with pid [10520]
[INFO] [spawner-8]: process started with pid [10522]
[INFO] [spawner-9]: process started with pid [10524]
[INFO] [spawner-10]: process started with pid [10526]
[INFO] [spawner-11]: process started with pid [10528]
[robot_state_publisher-5] [INFO] [1714472531.003165445] [robot_state_publisher]: got segment base
[robot_state_publisher-5] [INFO] [1714472531.003239721] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1714472531.003247353] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-5] [INFO] [1714472531.003251170] [robot_state_publisher]: got segment flange
[robot_state_publisher-5] [INFO] [1714472531.003255484] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-5] [INFO] [1714472531.003260135] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-5] [INFO] [1714472531.003265041] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-5] [INFO] [1714472531.003270075] [robot_state_publisher]: got segment tool0
[robot_state_publisher-5] [INFO] [1714472531.003275655] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-5] [INFO] [1714472531.003281201] [robot_state_publisher]: got segment world
[robot_state_publisher-5] [INFO] [1714472531.003286521] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-5] [INFO] [1714472531.003291766] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-5] [INFO] [1714472531.003297402] [robot_state_publisher]: got segment wrist_3_link
[dashboard_client-2] [INFO] [1714472531.008911335] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1714472531.014894320] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [WARN] [1714472531.017007124] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1714472531.019092681] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1714472531.022270611] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1714472531.022337316] [resource_manager]: Successful initialization of hardware 'ur5e'
[ur_ros2_control_node-1] [INFO] [1714472531.022546026] [resource_manager]: 'configure' hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1714472531.022556656] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1714472531.022569969] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1714472531.023522619] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ur_ros2_control_node-1] [INFO] [1714472531.253816623] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1714472531.254405652] [UR_Client_Library:]: Setting up RTDE communication with frequency 500.000000
[ur_ros2_control_node-1] [ERROR] [1714472532.278332632] [UR_Client_Library:]: Failed to initialize RTDE client, retrying in 10 seconds
[spawner-7] [INFO] [1714472533.471371367] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-10] [INFO] [1714472533.485289128] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-9] [INFO] [1714472533.499340380] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-11] [INFO] [1714472533.500807441] [spawner_forward_position_controller]: Waiting for '/controller_manager' services to be available
[... cut ...]
dpkg-buildpackage: Errore: debian/rules build subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:77: deb-pkg] Errore 2
make: *** [Makefile:1464: deb-pkg] Errore 2

I have tried something like commenting some config keys, as in this link [https://askubuntu.com/questions/1327749/error-in-compiling-installing-realtime-kernel-on-ubuntu-20-04-2-lts](url) , but it did not work.

I have chnaged my path, I am now trying to learn a little bit of the ur_driver, thinking it will help me to integrate the gripper to the ur.

fmauch, 2024-06-12:
I will close this issue, since there was a solution proposed roughly a month ago. Feel free to comment / reopen if there is still any issue persisting.

closed 2024-06-12 by fmauch