UniversalRobots/Universal_Robots_ROS2_Driver #983: [Humble] Gazebo: What is the intended way of setting up an UR10e

RobinHeitz, 2024-04-30:
### Affected ROS2 Driver version(s)

apt-release (ros-humle-ur)

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

No real robot so far

### How is the ROS driver used.

Others

### Issue details

# Summary
Based on documentation and launch file inspection of ur_moveit_config and ur_robot_driver, it's not clear to me how to use the ur in gazebo.

## Issue details
I've been using the ur_moveit_config and ur_robot_driver packages to have rviz and being able to plan & execute some movements (with use_fake_hardware = true).
I've spawned gazebo with `ros2 launch gazebo_ros gazebo.launch.py` to start the simulator.

Further, it's told by some tutorials, that the spawner can 'easily' introduce the robot based on /robot_description to the simulation scene. It runs successful, stating that the robot has been added. I can see it appeared under models, but is not visible.

I've seen that the urdf has some arg called sim_gazebo which false and cannot be accessed by the 2 launchfiles ur_moveit.launch.py and ur10e.launch.py. So I'm wondering, what is the intend way of making the robot visible in Gazebo? Or is it a completely other problem?

## Steps to Reproduce
Install ur packages with
apt install ros-humble-ur, execute the launch files mentioned above and then start gazebo. Try to add the robot to the scene with the spawner skript: `ros2 run gazebo_ros spawn_entity.py -topic robot_description -entity my_robot`
I would expect to see a robot in Gazebo. Although it's listened under the models section, its not visible.

I assume it has something to do with the argument in ur.urdf.xacro called 'sim_gazebo'.

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-05-02:
Did you take a look at https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/tree/humble?

VinDp, 2024-06-12:
Hey @RobinHeitz, 
since a solution has been proposed and some time has passed, I will close the issue.
Feel free to re-open if still relevant.

closed 2024-06-12 by VinDp