UniversalRobots/Universal_Robots_ROS2_Driver #1015: UR16e bringup in rviz2 laying down

eladpar, 2024-06-04:
### Affected ROS2 Driver version(s)

2.2.10

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

?

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
sometimes robot bring up finishes with robot laying down as described in the image
[image]

## Steps to Reproduce
1. launch robot driver
2. play program in robot
3. launch moveit config

## Expected Behavior
robot standing up and receiveng commands

## Actual Behavior
robot laying down

### Relevant log output

```shell
this is driver output:

source ~/ws_ur_moveit/install/setup.bash && ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur16e robot_ip:=192.168.1.102 launch_rviz:=false initial_joint_controller:=scaled_joint_trajectory_controller
[INFO] [launch]: All log files can be found below /home/hvmr/.ros/log/2024-06-04-12-32-01-377010-hvmr-NUC12DCMi9-16009
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [16014]
[INFO] [controller_stopper_node-3]: process started with pid [16016]
[INFO] [ur_ros2_control_node-1]: process started with pid [16012]
[INFO] [urscript_interface-4]: process started with pid [16018]
[INFO] [robot_state_publisher-5]: process started with pid [16020]
[INFO] [spawner-6]: process started with pid [16022]
[INFO] [spawner-7]: process started with pid [16024]
[INFO] [spawner-8]: process started with pid [16026]
[INFO] [spawner-9]: process started with pid [16028]
[INFO] [spawner-10]: process started with pid [16030]
[INFO] [spawner-11]: process started with pid [16032]
[dashboard_client-2] [INFO] [1717493521.757483772] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[ur_ros2_control_node-1] [WARN] [1717493521.760230004] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1717493521.760521349] [resource_manager]: Loading hardware 'ur16e' 
[controller_stopper_node-3] [INFO] [1717493521.760865729] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[... 149 lines cut ...]
[move_group-1] Loading 'move_group/MoveGroupMoveAction'...
[move_group-1] Loading 'move_group/MoveGroupPlanService'...
[move_group-1] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-1] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-1] 
[move_group-1] You can start planning now!
[move_group-1] 
[rviz2-2] [INFO] [1717493560.954811137] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.041605 seconds
[rviz2-2] [INFO] [1717493560.954901142] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[rviz2-2] [INFO] [1717493560.954927037] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-2] [INFO] [1717493563.311228336] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-2] [INFO] [1717493563.348797768] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-2] [INFO] [1717493564.194146871] [interactive_marker_display_108128111675392]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-2] [INFO] [1717493564.225167091] [interactive_marker_display_108128111675392]: Sending request for interactive markers
[rviz2-2] [INFO] [1717493564.278505930] [interactive_marker_display_108128111675392]: Service response received for initialization
[rviz2-2] [INFO] [1717493569.256020865] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: GetPlanningScene service call to get_planning_scene timed out. at ./planning_scene_monitor/src/planning_scene_monitor.cpp:562
[rviz2-2] [INFO] [1717493569.256079716] [moveit_ros_visualization.motion_planning_frame]: group ur_manipulator
[rviz2-2] [INFO] [1717493569.256085747] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'ur_manipulator' in namespace ''
[rviz2-2] [INFO] [1717493577.223360133] [move_group_interface]: Ready to take commands for planning group ur_manipulator.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

### Accept Public visibility

- [X] I agree to make this context public

VinDp, 2024-06-05:
Hi @eladpar, 
from you log output it seems like you are stopping `ur_control.launch.py` to start `ur_moveit.launch.py`, that wouldn't work because they need to be both active in different terminals. 
Assuming you are already doing that, the behavior you are describing sometimes happens when moveit starts and it still didn't receive any joint data, so it defaults to the zero configuration. 
It could be also related to the fact that controller_spawner doesn't come up for you and this could be due to [1483](https://github.com/ros-controls/ros2_control/pull/1483) still not being merged.

eladpar, 2024-06-06:
> Hi @eladpar, from you log output it seems like you are stopping `ur_control.launch.py` to start `ur_moveit.launch.py`, that wouldn't work because they need to be both active in different terminals. [...]

im using them in 2 different terminals, just pasted them together.

urfeex, 2025-01-15:
Closing as this seems not to be reproducible. Please reopen, if the problem still exists and you have further information on how to reproduce it.

closed 2025-01-15 by urfeex