UniversalRobots/Universal_Robots_ROS2_Driver #1032: Ros node unable to see robot_description_semantic parameter

Payday02, 2024-06-20:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

latest

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
I am trying to run the example ros2 moveit planning node to interact with ros_ur_driver's standard bringup and rviz setup, but the node is unable to find the parameters it is looking for and will not work. I have also tried this code with the standard panda arm demo and it works properly.

## Issue details
arm_server.cpp: 
```#include <memory>

#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char * argv[])
{
  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
    "arm_server",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );

  // Create a ROS logger
  auto const logger = rclcpp::get_logger("server_arm Initializing");

  // Create the MoveIt MoveGroup Interface
using moveit::planning_interface::MoveGroupInterface;
auto move_group_interface = MoveGroupInterface(node, "ur_manipulator");

// Set a target Pose
auto const target_pose = []{
  geometry_msgs::msg::Pose msg;
  msg.orientation.w = 1.0;
  msg.position.x = 0.28;
  msg.position.y = -0.2;
  msg.position.z = 0.5;
  return msg;
}();
move_group_interface.setPoseTarget(target_pose);

// Create a plan to that target pose
auto const [success, plan] = [&move_group_interface]{
  moveit::planning_interface::MoveGroupInterface::Plan msg;
  auto const ok = static_cast<bool>(move_group_interface.plan(msg));
  return std::make_pair(ok, msg);
}();

// Execute the plan
if(success) {
  move_group_interface.execute(plan);
} else {
  RCLCPP_ERROR(logger, "Planning failed!");
}

  // Shutdown ROS
  rclcpp::shutdown();
  return 0;
}
```

## Steps to Reproduce

1. Install Ur Ros Driver from source
2. In one terminal run `ros2 run ur_client_library start_ursim.sh -m ur3e`
3. Access the sim over web browser
4. In a second terminal run `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 launch_rviz:=false`
5. Make sure it launches properly, then program the arm to use external input and start that program
6. In a third terminal run `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true`
7. In a fourth terminal run the arm_server.cpp node `ros2 run arm_server arm_server`

## Expected Behavior
Expected the node to send a goal to moveit2 and attempt to plan and execute that goal on the simulated arm

## Actual Behavior
Node throws two errors and does not start properly because it cannot find the parameters it is looking for

### Relevant log output

```shell
$ ros2 run arm_server arm_server

[ERROR] [1718893245.617361543] [arm_server]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.

Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0

         at line 732 in /home/pdavis/ws_moveit2/src/srdfdom/src/model.cpp

[ERROR] [1718893245.903549444] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF

[FATAL] [1718893245.903736816] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.

terminate called after throwing an instance of 'std::runtime_error'

  what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.

[ros2run]: Aborted
```

### Accept Public visibility

- [X] I agree to make this context public

Payday02, 2024-06-20:
Another note, a python node using the moveit2 python api for the same thing exhibits the same behavior

Payday02, 2024-06-26:
For some reason the C++ node for moveit will work when handed the information through a launch file, even though the python will not. I don't believe this should be necessary overall since the demo nodes for moveit do not require this but as I have a workaround this is no longer a pressing issue for me.

fmauch, 2024-06-27:
As you wrote above, it is required to pass the respective parameters to the arm_server node, as well.

Since this is more of a MoveIt usage issue I'll close this issue. Please keep in mind that the moveit_config from this repo is merely an example. For real-world applications you should create your own moveit_config package containing your actual configuration.

closed 2024-06-27 by fmauch

mfacton, 2024-07-18:
I'm also facing this same issue

Payday02, 2024-07-18:
As mentioned above, the solution for me was to create a launchfile that passes the relevant information to a cpp node, even though this was not necessary for the moveit2 tutorial example. For me this didn't matter much since I was going to be creating a larger launchfile anyway to launch the driver, moveit, and a couple of custom nodes all at once. Here is an example launchfile that worked for me if this would help you.
```
import launch
import os
import sys

from launch_ros.actions import Node
from launch.substitutions import PathJoinSubstitution, Command, FindExecutable
from launch_ros.substitutions import FindPackageShare

def get_robot_description():
    joint_limit_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3e", "joint_limits.yaml"]
    )
    kinematics_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3e", "default_kinematics.yaml"]
    )
    physical_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3e", "physical_parameters.yaml"]
    )
    visual_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3e", "visual_parameters.yaml"]
    )
    robot_description_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution([FindPackageShare("ur_description"), "urdf", "ur.urdf.xacro"]),
            " ",
            "robot_ip:=172.17.0.2",
            " ",
            "joint_limit_params:=",
            joint_limit_params,
            " ",
            "kinematics_params:=",
            kinematics_params,
            " ",
            "physical_params:=",
            physical_params,
            " ",
            "visual_params:=",
            visual_params,
            " ",
           "safety_limits:=",
            "true",
            " ",
            "safety_pos_margin:=",
            "0.15",
            " ",
            "safety_k_position:=",
            "20",
            " ",
            "name:=",
            "ur",
            " ",
            "ur_type:=",
            "ur3e",
            " ",
            "prefix:=",
            '""',
            " ",
        ]
    )

    robot_description = {"robot_description": robot_description_content}
    return robot_description

def get_robot_description_semantic():
    # MoveIt Configuration
    robot_description_semantic_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution([FindPackageShare("ur_moveit_config"), "srdf", "ur.srdf.xacro"]),
            " ",
            "name:=",
            # Also ur_type parameter could be used but then the planning group names in yaml
            # configs has to be updated!
            "ur",
            " ",
            "prefix:=",
            '""',
            " ",
        ]
    )
    robot_description_semantic = {
        "robot_description_semantic": robot_description_semantic_content
    }
    return robot_description_semantic

def generate_launch_description():
    # generate_common_hybrid_launch_description() returns a list of nodes to launch
    robot_description = get_robot_description()
    robot_description_semantic = get_robot_description_semantic()
    demo_node = Node(
        package="arm_server",
        executable="arm_server",
        name="arm_server",
        output="screen",
        parameters=[
            robot_description,
            robot_description_semantic,
        ],
    )

    return launch.LaunchDescription([demo_node])
```