UniversalRobots/Universal_Robots_ROS2_Driver #1035: UR Client Library fails to read from stream

Payday02, 2024-06-26:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

latest

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Though I can ping my robot at the IP address I specify, the ros drivers are unable to correctly read the stream from the robot and therefore will not let me send movement commands from moveit2.
## Issue details
My UR3e is connected to the computer through a ethernet to usb-c dongle, which is then routed into my VM running ubuntu 22.04. I am able to launch the drivers with URSim and successfully plan movements with moveit2, but run into this issue when I try to connect the drivers with my physical arm. I have tested and I am able to ping the arm at various IPs set in the teach pendant though the robot always says it is Not connected to network even though the static address settings work. I have ensured that all utilized ports are not in use by anything else and moveit is receiving robot position data so some communication is happening. I am also able to see the ready to receive commands message when I start the URCap external control program on the teach pendant but it throws the following when I try to then send a command:
```
[ur_ros2_control_node-1] [INFO] [1719421979.084573597] [scaled_joint_trajectory_controller]: Received new action goal

[ur_ros2_control_node-1] [ERROR] [1719421979.085979051] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running
```
## Steps to Reproduce
Connect robot to computer as described above
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.100.101 launch_rviz:=false`

## Expected Behavior
Normal driver activation

## Actual Behavior
Failed to read from stream, reconnecting in X seconds... messages as seen in log output

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/pdavis/.ros/log/2024-06-26-12-04-46-040930-FMTenv-5313

[INFO] [launch]: Default logging verbosity is set to INFO

[INFO] [dashboard_client-2]: process started with pid [5318]

[INFO] [controller_stopper_node-3]: process started with pid [5320]

[INFO] [ur_ros2_control_node-1]: process started with pid [5316]

[INFO] [urscript_interface-4]: process started with pid [5322]

[INFO] [robot_state_publisher-5]: process started with pid [5324]

[INFO] [spawner-6]: process started with pid [5326]

[INFO] [spawner-7]: process started with pid [5328]

[... 179 lines cut ...]

[ur_ros2_control_node-1] [WARN] [1719421494.749606715] [UR_Client_Library:]: Failed to read from stream, reconnecting in 2 seconds...

[spawner-7] [INFO] [1719421494.854362853] [spawner_joint_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster

[ur_ros2_control_node-1] [INFO] [1719421495.085332625] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'

[spawner-7] [INFO] [1719421495.469109334] [spawner_joint_state_broadcaster]: Loaded force_torque_sensor_broadcaster

[ur_ros2_control_node-1] [INFO] [1719421495.623403010] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'

[spawner-7] [INFO] [1719421495.797414056] [spawner_joint_state_broadcaster]: Configured and activated force_torque_sensor_broadcaster

[INFO] [spawner-7]: process has finished cleanly [pid 5328]

[ur_ros2_control_node-1] [WARN] [1719421497.774811592] [UR_Client_Library:]: Failed to read from stream, reconnecting in 4 seconds...

[ur_ros2_control_node-1] [WARN] [1719421502.802582755] [UR_Client_Library:]: Failed to read from stream, reconnecting in 8 seconds...

[ur_ros2_control_node-1] [WARN] [1719421511.828092935] [UR_Client_Library:]: Failed to read from stream, reconnecting in 16 seconds...
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-06-27:
Running the driver in a VM can significantly slow down things / increase delays. I'm not saying, it is not possible to get this somehow reliable, but I expect that this would need some tweaking. It will also be very important **how** the network adapter is routed to the VM. If possible, I would suggest making the physical NIC available in the VM to reduce possible delays at this stage.

urfeex, 2025-02-06:
Closing this due to inactivity. Feel free to comment / reopen if there are any updates also considering the last comment.

closed 2025-02-06 by urfeex