UniversalRobots/Universal_Robots_ROS2_Driver #1062: Build from source is broken for Humble by latest updates in ros2_controllers package

cosmac, 2024-07-17:
### Affected ROS2 Driver version(s)

2.2.14

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.1

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
Failed to build Humble branch from source. External function signature changed (see `get_segment_tolerances()` from [ros2-controllers](https://github.com/ros-controls/ros2_controllers/blob/humble/joint_trajectory_controller/include/joint_trajectory_controller/tolerances.hpp#L96)).

## Issue details
- Please see https://github.com/ros-controls/ros2_controllers/commit/112df35d9b1fd4353035a9bbac1c1a256b7eec15 for more details on recent changes to ros2-controllers

## Steps to Reproduce
```
git clone --branch humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.humble.repos

colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```

## Expected Behavior
Build from source successful for all packages.

## Actual Behavior
Build fails during `ur_controllers`

# Workaround Suggestion
Update function call:
`default_tolerances_ = get_segment_tolerances(get_node()->get_logger(), params_);`

### Relevant log output

```shell
171.3 --- stderr: ur_controllers
171.3 /root/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function 'virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)':
171.3 /root/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:92:49: error: no matching function for call to 'get_segment_tolerances(joint_trajectory_controller::Params&)'
171.3    92 |     default_tolerances_ = get_segment_tolerances(params_);
171.3       |                           ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
171.3 In file included from /root/workspace/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:31,
171.3                  from /root/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
171.3                  from /root/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
171.3 /root/workspace/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:96:19: note: candidate: 'joint_trajectory_controller::SegmentTolerances joint_trajectory_controller::get_segment_tolerances(rclcpp::Logger&, const Params&)'
171.3    96 | SegmentTolerances get_segment_tolerances(rclcpp::Logger & jtc_logger, const Params & params)
171.3       |                   ^~~~~~~~~~~~~~~~~~~~~~
171.3 /root/workspace/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:96:19: note:   candidate expects 2 arguments, 1 provided
171.3 /root/workspace/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:166:19: note: candidate: 'joint_trajectory_controller::SegmentTolerances joint_trajectory_controller::get_segment_tolerances(rclcpp::Logger&, const SegmentTolerances&, const control_msgs::action::FollowJointTrajectory::Goal&, const std::vector<std::__cxx11::basic_string<char>, std::allocator<std::__cxx11::basic_string<char> > >&)'
171.3   166 | SegmentTolerances get_segment_tolerances(
171.3       |                   ^~~~~~~~~~~~~~~~~~~~~~
171.3 /root/workspace/ros_ur_driver/install/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:166:19: note:   candidate expects 4 arguments, 1 provided
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-07-18:
Yes, that needs another round of refactoring. Thanks for pointing that out, though.

utsavrai, 2024-07-26:
Hi @fmauch, the build is still failing for humble.

```
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> ur_dashboard_msgs
Starting >>> ur_description
Finished <<< ur_description [0.80s]                                                               
Starting >>> ur_moveit_config                         
Finished <<< ur_moveit_config [0.56s]                                                                    
Finished <<< ur_dashboard_msgs [2.04s]                     
Starting >>> ur_controllers
--- stderr: ur_controllers                              
/home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:93:50: error: invalid initialization of reference of type ‘const joint_trajectory_controller::Params&’ from expression of type ‘rclcpp::Logger’
   93 |     default_tolerances_ = get_segment_tolerances(logger, params_);
      |                                                  ^~~~~~
In file included from /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:31,
                 from /home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:91:57: note: in passing argument 1 of ‘joint_trajectory_controller::SegmentTolerances joint_trajectory_controller::get_segment_tolerances(const joint_trajectory_controller::Params&)’
   91 | SegmentTolerances get_segment_tolerances(Params const & params)
      |                                          ~~~~~~~~~~~~~~~^~~~~~
/home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:184:25: error: ‘active_tolerances_’ was not declared in this scope
  184 |       auto active_tol = active_tolerances_.readFromRT();
      |                         ^~~~~~~~~~~~~~~~~~
/home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:204:14: error: ‘check_state_tolerance_per_joint’ was not declared in this scope; did you mean ‘joint_trajectory_controller::check_state_tolerance_per_joint’?
  204 |             !check_state_tolerance_per_joint(state_error_, index, active_tol->state_tolerance[index],
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              joint_trajectory_controller::check_state_tolerance_per_joint
In file included from /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:31,
                 from /home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:126:13: note: ‘joint_trajectory_controller::check_state_tolerance_per_joint’ declared here
  126 | inline bool check_state_tolerance_per_joint(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:210:14: error: ‘check_state_tolerance_per_joint’ was not declared in this scope; did you mean ‘joint_trajectory_controller::check_state_tolerance_per_joint’?
  210 |             !check_state_tolerance_per_joint(state_error_, index, active_tol->goal_state_tolerance[index],
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              joint_trajectory_controller::check_state_tolerance_per_joint
In file included from /opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/joint_trajectory_controller.hpp:31,
                 from /home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/scaled_joint_trajectory_controller.hpp:41,
                 from /home/utsav/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:41:
/opt/ros/humble/include/joint_trajectory_controller/joint_trajectory_controller/tolerances.hpp:126:13: note: ‘joint_trajectory_controller::check_state_tolerance_per_joint’ declared here
  126 | inline bool check_state_tolerance_per_joint(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:223: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [8.41s, exited with code 2]

Summary: 3 packages finished [10.6s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  4 packages not processed

```

fmauch, 2024-07-26:
Yes, this is normal and expected. The upstream changes showed up in the semi-binary builds as soon as they hit the humble branch upstream. Now, that API is published as package to the testing repo, hence the testing binary build was also failing, that was when this issue was created. Since there has not been a sync from testing to main, main still contains the old API, so now the main binary build is failing. This will only affect users building this repo from source without building ros2_controllers from source. There is no way around this, when upstream API is changed.

utsavrai, 2024-07-26:
@fmauch Thanks for answering. I wanted to understand how to successfully build the humble branch from source successfully as I have pulled all the changes and I am still failing to build.  Are there more fixes to come?

Thanks

fmauch, 2024-07-26:
As I've written, for the current humble branch to build you'll need to build ros2_control (and therefore moveit) from source. Or you wait for the next humble sync ;-)

Unless you want to actively take part in development, there should be no reason to build this package from source.

utsavrai, 2024-07-26:
Understood. Thank you very much for responding!

PedroMoreo, 2024-08-02:
I have the same problem.
Thanks a lot.

fmauch, 2024-08-08:
The humble sync is out, all humble pipelines are green, that should be resolved.

closed 2024-08-08 by fmauch