UniversalRobots/Universal_Robots_ROS2_Driver #1070: Exception caught while processing action 'loadRobotModel'

Testeh69, 2024-08-02:
### Affected ROS2 Driver version(s)

Jazzy

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.14.1

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I try to launch move it with the launch file and get an error

## Issue details
[rviz2_moveit.BackgroundProcessing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.shoulder_pan_joint.max_acceleration' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.shoulder_pan_joint.max_acceleration} is of type {double}, setting it to {string} is not allowed.

## Steps to Reproduce
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=169.254.215.36 launch_rviz:=false
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true

## Expected Behavior
Rviz opening with moveit correctly

## Actual Behavior
RViz opening but no robot appears and i get the error mentionned

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-08-07:
This looks like https://github.com/moveit/moveit2/issues/1882

To workaround this, use `LC_ALL=C` in front of your launch call:
```
LC_ALL=C ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true
```

VinDp, 2024-09-18:
Since a solution has been proposed and some time has passed, I will close this issue.
Feel free to reopen if still relevant.

closed 2024-09-18 by VinDp