UniversalRobots/Universal_Robots_ROS2_Driver #1072: Trajectory execution failed with error: PATH_TOLERANCE_VIOLATED

Yazan-Soliman, 2024-08-06:
### Affected ROS2 Driver version(s)

2.2.14-1

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot, Real robot, URSim in docker

### Robot SW / URSim version(s)

5.16.0

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
moving the robot (ur10e) over scaled_joint_trajectory_controller  when in using ursim docker or on the real robot (ur10e)  returning the following error

`[ERROR] [1722985034.879335744] [ur10e_trajectory_planner]: Trajectory execution failed with error: PATH_TOLERANCE_VIOLATED`

## Issue details
*Detailed description help us understand the problem. Code are welcome!*

## Steps to Reproduce
launch the ur_robot_driver to a real robot or a ursim docker
execute test trajectory like in 
ur_robot_driver/scripts/example_move.py

## Expected Behavior
when only using rviz with use?fake_hardware:= the robot follows the desired trajectories.
but when connected to the real robot or ursim docker the above error occurs.

## Actual Behavior
robot does not move

### Relevant log output

```shell
`[ur_ros2_control_node-1] [INFO] [1722985006.778117990] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1722985006.778220070] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-1] [WARN] [1722985034.829688924] [scaled_joint_trajectory_controller]: Aborted due to state tolerance violation
`
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-08-07:
What kind of trajectory are you sending to the robot? Is it unfeasible in any way? Does using the `example_move.py` script run without errors?

Yazan-Soliman, 2024-08-07:
> What kind of trajectory are you sending to the robot? Is it unfeasible in any way? Does using the `example_move.py` script run without errors?

The trajectories are feasible from what I see when they are executed when use_fake_hardware:=true on rviz, on the real hardware unfortunately  not. 
`example_move.py` shows the same error 

`[INFO] [1723022412.532784810] [jtc_client]: Waiting for action server on scaled_joint_trajectory_controller/follow_joint_trajectory
[INFO] [1723022412.533669464] [jtc_client]: Executing trajectory traj0
[INFO] [1723022413.086584979] [jtc_client]: Done with result: PATH_TOLERANCE_VIOLATED
Traceback (most recent call last):
  File "Universal_Robots_ROS2_Driver/ur_robot_driver/scripts/example_move.py", line 190, in <module>
    main()
  File "Universal_Robots_ROS2_Driver/ur_robot_driver/scripts/example_move.py", line 182, in main
    rclpy.spin(jtc_client)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
    executor.spin_once()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 739, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 736, in _spin_once_impl
    raise handler.exception()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 251, in __call__
    self.set_result(self._handler(*self._args, **self._kwargs))
  File "Universal_Robots_ROS2_Driver/ur_robot_driver/scripts/example_move.py", line 159, in get_result_callback
    raise RuntimeError("Executing trajectory failed")
RuntimeError: Executing trajectory failed`

from the running terminal

`[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ur_ros2_control_node-1]: process started with pid [95974]
[INFO] [robot_state_publisher-2]: process started with pid [95976]
[INFO] [rviz2-3]: process started with pid [95978]
[INFO] [spawner-4]: process started with pid [95980]
[INFO] [spawner-5]: process started with pid [95982]
[INFO] [spawner-6]: process started with pid [95984]
[INFO] [spawner-7]: process started with pid [95986]
[INFO] [spawner-8]: process started with pid [95988]
[INFO] [spawner-9]: process started with pid [95990]
[ur_ros2_control_node-1] [WARN] [1723021998.289086273] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-1] text not specified in the tf_prefix tag
[ur_ros2_control_node-1] [INFO] [1723021998.289590025] [resource_manager]: Loading hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1723021998.292222499] [resource_manager]: Initialize hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1723021998.292397606] [resource_manager]: Successful initialization of hardware 'ur10e'
[ur_ros2_control_node-1] [INFO] [1723021998.292634663] [resource_manager]: 'configure' hardware 'ur10e' 
[ur_ros2_control_node-1] [INFO] [1723021998.292658973] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1723021998.292679962] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1723021998.293509758] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[robot_state_publisher-2] [INFO] [1723021998.294664304] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1723021998.295051369] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1723021998.295088458] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1723021998.295098827] [robot_state_publisher]: got segment cell_base_link
[robot_state_publisher-2] [INFO] [1723021998.295105588] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1723021998.295111816] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1723021998.295120256] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1723021998.295126142] [robot_state_publisher]: got segment robot_mount_link
[robot_state_publisher-2] [INFO] [1723021998.295131539] [robot_state_publisher]: got segment second_link_1
[robot_state_publisher-2] [INFO] [1723021998.295138156] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1723021998.295144374] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1723021998.295150344] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1723021998.295156881] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1723021998.295163184] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1723021998.295169362] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1723021998.295175492] [robot_state_publisher]: got segment wrist_3_link
[rviz2-3] [INFO] [1723021998.835684166] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1723021998.836061367] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ur_ros2_control_node-1] [INFO] [1723021998.857990575] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
[... cut ...]
[ur_ros2_control_node-1] [WARN] [1723022301.602219145] [scaled_joint_trajectory_controller]: Aborted due to state tolerance violation
[ur_ros2_control_node-1] [INFO] [1723022412.534407237] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1723022412.534483626] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-1] [WARN] [1723022413.050250036] [scaled_joint_trajectory_controller]: Aborted due to state tolerance violation
`

Yazan-Soliman, 2024-08-08:
I managed to solve the problem, it was a firewall problem!!!  I am not sure how to relate the error of state tolerance violation to it but what hinted it to me was testing from a different machine (laptop) and noticing in the log
```[ur_ros2_control_node-1] [INFO] [1723119165.138680125] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.```

which i did not get on the my laptop. 

```[ur_ros2_control_node-1] [WARN] [1723022413.050250036] [scaled_joint_trajectory_controller]: Aborted due to state tolerance violation``` and this was very misleading to begin with.

Yazan-Soliman, 2024-08-08:
I managed to solve the problem – it was a firewall issue! I'm still not entirely sure how the error of state tolerance violation relates to the firewall, but here's what tipped me off:

Testing on a different laptop I noticed the following log entry:
```[ur_ros2_control_node-1] [INFO] [1723119165.138680125] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.```
I did not see this log entry on my primary machine, also 
```[ur_ros2_control_node-1] [WARN] [1723022413.050250036] [scaled_joint_trajectory_controller]: Aborted due to state tolerance violation```  was very misleading initially.

VinDp, 2024-09-18:
Since the issue seems to have been solved, I will close it.
Feel free to reopen if still relevant.

closed 2024-09-18 by VinDp