UniversalRobots/Universal_Robots_ROS2_Driver #1079: The connection to the remote PC could not be established, Reason: connect time out.

NorthWolf222, 2024-08-09:
### Affected ROS2 Driver version(s)

git clone -b humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR CB3 robot, Real robot

### Robot SW / URSim version(s)

I do not know

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*
    Affected UR Software Version(s): 3.14.1
    Affected Robot Hardware Version(s): UR5
    URCaps Software version(s): external control 1.0.1

My computer is Ubuntu 22.04, ROS-Humble, After I enabled urcap in the teach pendant, It will pop up with the following error: 
[image]
My PC's ros2_drivers seems to start up normally, but the URcap on the real robot arm UR5 is unable to establish a connection with the PC.*

## Issue details
*When I run the launch file, the result is as follows:
shiou@wsoPC:~/UR5_Driver_WS$ ros2 launch ur_robot_driver ur5.launch.py robot_ip:=192.168.1.101 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/shiou/.ros/log/2024-08-09-17-15-20-523833-wsoPC-29244
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [29249]
[INFO] [controller_stopper_node-3]: process started with pid [29251]
[INFO] [ur_ros2_control_node-1]: process started with pid [29247]
[INFO] [urscript_interface-4]: process started with pid [29253]
[INFO] [robot_state_publisher-5]: process started with pid [29255]
[INFO] [rviz2-6]: process started with pid [29257]
[INFO] [spawner-7]: process started with pid [29259]
[INFO] [spawner-8]: process started with pid [29261]
[INFO] [spawner-9]: process started with pid [29263]
[dashboard_client-2] [INFO] [1723194924.647880800] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1723194924.789804173] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-5] [INFO] [1723194925.486561503] [robot_state_publisher]: got segment base
[robot_state_publisher-5] [INFO] [1723194925.487300465] [robot_state_publisher]: got segment base_link
[robot_state_publisher-5] [INFO] [1723194925.487388370] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-5] [INFO] [1723194925.487424695] [robot_state_publisher]: got segment flange
[robot_state_publisher-5] [INFO] [1723194925.487455522] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-5] [INFO] [1723194925.487484975] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-5] [INFO] [1723194925.487516618] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-5] [INFO] [1723194925.487546504] [robot_state_publisher]: got segment tool0
[robot_state_publisher-5] [INFO] [1723194925.487575596] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-5] [INFO] [1723194925.487604354] [robot_state_publisher]: got segment world
[robot_state_publisher-5] [INFO] [1723194925.487633541] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-5] [INFO] [1723194925.487663094] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-5] [INFO] [1723194925.487691877] [robot_state_publisher]: got segment wrist_3_link
[... 61 lines cut ...]
[rviz2-6] [INFO] [1723194932.129054424] [rviz2]: Stereo is NOT SUPPORTED

and the IP settings are as follows:
[image]

[image]

[image]

But when I run the external control program on the teach pendant side, the following appears:
[image]

I tried to solve the issue by turning off the firewall, but it didn't work, and the connection timeout problem still occurs.

The 192.168.1.101 can be pinged successfully. And when I run the launch file, run the following command:
[image]

It seems that port 50002 is open, but I don't know why running URCap shows a connection timeout.

What could be causing this and how could I fix it? Any help would be highly appreciated. Thanks
*

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-09-04:
Thank you for the in-detail report. From what I can see, everything looks correct, so I don't have any direct advice, I'm afraid.

> I tried to solve the issue by turning off the firewall, but it didn't work, and the connection timeout problem still occurs.

That's the only point where I can imagine things going wrong. On ubuntu there are multiple possibilities for having a firewall. Which ones did you check for?

optimusprime222, 2024-11-03:
Thank you for your reply. I couldn't find a suitable solution either. Later, I successfully configured it using the same method on another computer, which was amazing and speechless. Perhaps my previous laptop model was older.

urfeex, 2025-01-15:
Closing this as the problem has been solved / workarounded by the OP and there is no general or specific error to track / fix.