UniversalRobots/Universal_Robots_ROS2_Driver #1083: using moveit computeTimeStamps uses maximum velocity and acceleration values that are different from those used by computeCartesianPath

bodkal, 2024-08-11:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

ros2 run ur_client_library start_ursim.sh -m ur10e

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
Hi all,

I'm using a UR10e robot and trying to control the Cartesian path's velocity and acceleration through code.

When I generate a Cartesian path, it uses the default values from the ur_moveit_config/config/joint_limits.yaml file. To slow it down, I use the computeTimeStamps function with a scale of 1 and 1 (which isn't supposed to change anything), but I end up with a new scale with a maximum acceleration of 1.

Do computeTimeStamps and computeCartesianPath use the same joint_limits.yaml file? Any ideas?

related to issue #966

Thanks!

## Issue details
*Detailed description help us understand the problem. Code are welcome!*

## Steps to Reproduce

snip of code: 

`    double fraction = this->move_group_interface.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectory, false);

    // The trajectory needs to be modified so it will include velocities as well.

    // First to create a RobotTrajectory object

    robot_trajectory::RobotTrajectory rt(move_group_interface.getCurrentState()->getRobotModel(), "ur_manipulator");

    // Second get a RobotTrajectory from trajectory

    rt.setRobotTrajectoryMsg(*move_group_interface.getCurrentState(), trajectory);

    this->printRobotTrajectory(trajectory);

    // Thrid create a IterativeParabolicTimeParameterization object

    trajectory_processing::IterativeParabolicTimeParameterization iptp;

    // Fourth compute computeTimeStamps

[... 262 lines cut ...]
  wrist_3_joint:

    has_acceleration_limits: false

    has_effort_limits: true

    has_position_limits: true

    has_velocity_limits: true

    max_effort: 56.0

    max_position: !degrees  360.0

    max_velocity: !degrees  180.0

    min_position: !degrees -360.0
`

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-08-12:
This seems to be rather a moveit question than a question to this driver. If there is anything we can improve with our example moveit config, please let us know otherwise I am afraid, I will have to tell you, that moveit interaction does not lie inside our expertise and support range of this repository.

VinDp, 2024-09-18:
Since there was no update, I will close this issue.
Feel free to reopen if still relevant.

closed 2024-09-18 by VinDp