UniversalRobots/Universal_Robots_ROS2_Driver #1090: scaled_joint_trajectory_controller starts as inactive

pep248, 2024-08-29:
### Affected ROS2 Driver version(s)

latest

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

latest

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
scaled_joint_trajectory_controller starts as inactive

## Issue details
i ran the following commands in two different windows:
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=172.16.7.75 launch_rviz:=false`
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10e launch_rviz:=true`

and, by default I was unable to move the robot using the rviz MotionPlanning interface.

Then, on a 3rd terminal, i ran the following command:

`ros2 control list_controllers`

and the output was the following:

```
io_and_status_controller[ur_controllers/GPIOController] active    
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] inactive  
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[position_controllers/JointGroupPositionController] inactive  
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
```
I don't know why said controller is automatically set to inactive.

if I manually set it to active, it works like a charm:
`ros2 control set_controller_state scaled_joint_trajectory_controller active`
but I don't want to be doing this by hand.

I will share the output of the controller launch.

## Steps to Reproduce
I simply downloaded the ros-humble-ur-* packages and launched as described. 

## Expected Behavior
I expected the scaled_joint_trajectory_controller to start as "active"

## Actual Behavior
The scaled_joint_trajectory_controller to start as "inactive"
[... 81 lines cut ...]
[ur_ros2_control_node-1] [INFO] [1724923996.856469714] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ur_ros2_control_node-1] [INFO] [1724923996.860019028] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ur_ros2_control_node-1] [INFO] [1724923996.864997104] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1724923996.865045334] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ur_ros2_control_node-1] [INFO] [1724923996.867232910] [controller_manager]: Configuring controller 'forward_position_controller'
[ur_ros2_control_node-1] [INFO] [1724923996.867892209] [forward_position_controller]: configure successful
[ur_ros2_control_node-1] [INFO] [1724923996.869152246] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[ur_ros2_control_node-1] [INFO] [1724923996.871160008] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1724923996.871195454] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[spawner-8] [INFO] [1724923996.885431360] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[spawner-6] [INFO] [1724923996.889335615] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[spawner-7] [INFO] [1724923996.893659342] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[spawner-10] [INFO] [1724923996.897505679] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[spawner-9] [INFO] [1724923996.901391720] [spawner_speed_scaling_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[INFO] [spawner-11]: process has finished cleanly [pid 201]
[INFO] [spawner-8]: process has finished cleanly [pid 195]
[INFO] [spawner-6]: process has finished cleanly [pid 191]
[INFO] [spawner-9]: process has finished cleanly [pid 197]
[INFO] [spawner-7]: process has finished cleanly [pid 193]
[INFO] [spawner-10]: process has finished cleanly [pid 199]
```

### Accept Public visibility

- [X] I agree to make this context public

VinDp, 2024-09-04:
Hey @pep248, 
do you start the External Control program on the robot? Because without that the controller will stay inactive.
If you want, you can start it also from the terminal, using the `headless_mode` option:  `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=172.16.7.75 launch_rviz:=false headless_mode:=true`

pep248, 2024-09-04:
Thanks a lot! That did the trick!
[image]

closed 2024-09-04 by pep248