UniversalRobots/Universal_Robots_ROS2_Driver #1097: Failure During Hardware Component Reconfiguration

sahand-ghaffari-ocado, 2024-09-09:
### Affected ROS2 Driver version(s)

2.2.14

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.17.0

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Problem Description

Reconfiguring the UR hardware component (which can be done using the `~/set_hardware_component_state` service call to unconfigure and then activate the UR hardware component) results in a code failure due to following issues

1. The `registerUrclLogHandler()` [function](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_robot_driver/src/urcl_log_handler.cpp#L82) which is called in the  `on_configure()` [function](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_robot_driver/src/hardware_interface.cpp#L421) uses `std::move` to transfer ownership of `g_log_handler`. As a result, calling the `on_configure() `function again causes a failure. This is because the `g_log_handler` pointer has already been moved, and it no longer has access to the `setTFPrefix()` function, ultimately leading to a code failure.
2. The `checkAsyncIO()` [function](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_robot_driver/src/hardware_interface.cpp#L518) cannot be called in the` asyncThread() `function after reconfiguring the hardware component. This is because the `async_thread_shutdown_` variable is set to true in the `on_cleanup()` [function](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_robot_driver/src/hardware_interface.cpp#L477) and is not reset to false in the `on_configure()` function.

# Workaround Suggestion
To avoid such crashes, following changes are suggested

1. It is suggested to declare and assign the pointer within the `registerUrclLogHandler()` function instead of doing so globally. Therefore, it is suggested to modify the function as follows:

```
void registerUrclLogHandler(const std::string& tf_prefix)
{
  if (g_registered == false) {
    std::unique_ptr<UrclLogHandler> g_log_handler(new UrclLogHandler);
    g_log_handler->setTFPrefix(tf_prefix);
    // Log level is decided by ROS2 log level
    urcl::setLogLevel(urcl::LogLevel::DEBUG);
    urcl::registerLogHandler(std::move(g_log_handler));
    g_registered = true;
  }
}
```
2. It is suggested to set `async_thread_shutdown_` to `false` in `on_configure() `function

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-09-11:
Thank you for reporting this and already suggesting a way of resolving the issue.

To implement this it would be nice to have a test case actually covering this, though. I'll add a draft PR with the changes you suggested but before merging I would like to have tests for changing the lifecycle state. Unfortunately, I don't know how much time I will have available to spend on this.

sahand-ghaffari-ocado, 2024-09-16:
Thank you for your quick response and for moving forward with the draft PR! I agree that having a test case to cover the changes would be ideal before merging. What kind of tests are you thinking of adding to cover the lifecycle state change? I'd be happy to assist with writing or reviewing the tests if needed, especially if you're short on time. By the way, I also opened a [PR](https://github.com/ros-controls/ros2_control/pull/1755) in `ros2_control` to fix the bug related to hardware component unconfiguration.

fmauch, 2024-09-18:
If you could help out writing tests for this, that would be great!

Basically, I think a short integration test that starts the robot, cycles the states down and up once more, making sure the available interfaces match the lifecycle state on the way.

closed 2024-10-21 by VinDp

fmauch, 2024-10-22:
For reference: I created a separate issue regarding the tests: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1160