UniversalRobots/Universal_Robots_ROS2_Driver #1100: Follow On: How to boot multiple robots utilizing a single description without using ur_control.launch.py?

lcbw, 2024-09-12:
### Affected ROS2 Driver version(s)

Humble Branch

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

UR5e Polyscope 5.15, UR10e Polyscope 5.8

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
We are attempting to control a UR5e and a UR10e simultaneously with a singular description file and a singular controller manager through the external control URCap, which has been installed on each teach pendant. 
When launching a file that includes the description and port connection information for both robots, not all of the controllers start up because the hardware interfaces for the first robot are labelled as [unavailable] [unclaimed]. 

## Issue details
Each of the robots can be controlled separately to execute a simple trajectory using a joint_trajectory_controller with the same launch file and xacro, with the 'single_arm' argument set to true. 
On the teach pendant, each robot has the external control urcap ver. 1.0.5 installed and takes commands from ROS in remote mode. 
The robots IP addresses and port numbers are different and are constant throughout testing each robot individually and together. 

However, when my launch file (included below in repeat instructions) is ran the robots do not follow the trajectory provided and the following behaviors also occur: 
1. The joint_trajectory_controllers fails to initialize
2. The UR10e teach pendant shows "running" under the external control URCap in Remote mode as if the robot is connected to the computer correctly, and yet the UR5e teach pendant does not show "running".
3. Only the current pose of the UR5e appears in RVIZ. (The robot does not move either in sim or real life, because the joint traj controller fails to initialize)
4. The command "ros2 control list_hardware_interfaces" produces the following result (truncated for clarity):
command interfaces
	robot1/elbow_joint/position [**unavailable**] [**unclaimed**]
	robot1/elbow_joint/velocity [unavailable] [unclaimed]
	robot1/shoulder_lift_joint/position [unavailable] [unclaimed]
	robot1/shoulder_lift_joint/velocity [unavailable] [unclaimed]
..... and so on and so forth for all robot1 interfaces 
	robot2/elbow_joint/position [**available**] [**unclaimed**]
	robot2/elbow_joint/velocity [available] [unclaimed]
...... and so on and so forth for all robot2 interfaces 

## Steps to Reproduce
Please see the controllers.yaml, xacro, and launch files here:  https://github.com/lcbw/dual_arm_test

Then, having installed the UR driver, controllers and ROS2 humble from debians and confirming successful control of each robot using the External Control URCap, run the launch file. 

Only the joint_state_broadcaster should activate, and nothing should move. 

## Expected Behavior
I expected all controllers to configure, load, and activate and for the robots to follow the trajectory provided. 

## Actual Behavior
Only the joint_state_broadcaster should activate, nothing should move, and only the UR5e should be visible in RVIZ, and it will not move. 

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*
I have not found a workaround and would appreciate any insight anyone has on this end. Thank you! 
[... 139 lines cut ...]
[ros2_control_node-1]  robot1/shoulder_lift_joint/velocity
[ros2_control_node-1]  robot1/elbow_joint/position
[ros2_control_node-1]  robot1/elbow_joint/velocity
[ros2_control_node-1]  robot1/wrist_1_joint/position
[ros2_control_node-1]  robot1/wrist_1_joint/velocity
[ros2_control_node-1]  robot1/wrist_2_joint/position
[ros2_control_node-1]  robot1/wrist_2_joint/velocity
[ros2_control_node-1]  robot1/wrist_3_joint/position
[ros2_control_node-1]  robot1/wrist_3_joint/velocity
[ros2_control_node-1] ]
[ros2_control_node-1] 
[ros2_control_node-1] [ERROR] [1726174534.889949319] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[spawner-7] [ERROR] [1726174534.891071408] [spawner_joint_trajectory_controller]: Failed to activate controller
[ERROR] [spawner-7]: process has died [pid 1632321, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_trajectory_controller -c /controller_manager --ros-args'].
[INFO] [trajectory_loader.py-8]: process started with pid [1632396]
[trajectory_loader.py-8] [INFO] [1726174535.818930392] [motion_plan_from_file]: Waiting for action server on /joint_trajectory_controller/follow_joint_trajectory
[trajectory_loader.py-8] [INFO] [1726174535.819916636] [motion_plan_from_file]: Executing trajectory
[ros2_control_node-1] [INFO] [1726174535.820836182] [joint_trajectory_controller]: Received new action goal
[ros2_control_node-1] [ERROR] [1726174535.820903036] [joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
[trajectory_loader.py-8] [INFO] [1726174535.822086148] [motion_plan_from_file]: Goal rejected :(
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-09-18:
Hi there, thanks for reaching out! If you could change your linked example to be a usable minimal working example this would greatly help in debugging this from our side.

Looking at your log output I can see 
```
[ros2_control_node-1] [FATAL] [1726174534.067706847] [URPositionHardwareInterface]: Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured
[ros2_control_node-1] [INFO] [1726174534.067756095] [resource_manager]: Failed to 'configure' hardware 'ur10e'
```

which looks like https://github.com/UniversalRobots/Universal_Robots_ROS_Driver?tab=readme-ov-file#when-starting-the-driver-it-crashes-with-variable-speed_slider_mask-is-currently-controlled-by-another-rtde-client

Since the ur10e (robot1) doesn't come up, its interfaces aren't available, hence the controller cannot get activated.

Could you try fixing the ur10e startup and see if that helps?

lcbw, 2024-09-18:
Hi there, thank you for your prompt response! 

I have just updated https://github.com/lcbw/dual_arm_test/tree/main to be build and tested more easily. 

I have checked both the UR10e and the UR5e and both of them have always had the ethernet/ip adapter disabled. Did I misunderstand your recommendation for a solution?

lcbw, 2024-09-24:
I'm going to close this issue in favor of opening two separate issues that are more focused on the multiple issues I've encountered. 
Issue  - #1111  and #1112

closed 2024-09-24 by lcbw