UniversalRobots/Universal_Robots_ROS2_Driver #1109: Gripper connected to control box not connected

Mokchanic, 2024-09-23:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

URSoftware 3.14

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Gripper connected to control box not connected.

## Issue details
Gripper is recognized when directly connected to the computer, but not in the control box.
I referred to the link below, but it doesn't seem to work in cb3 because it's e-series only.

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_robot_driver/doc/setup_tool_communication.rst

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-10-09:
Yes, this does not work.

q434343, 2024-10-10:
> Yes, this does not work.

thanks! 
I met the same problem and I now find out that 'use_tool_communication' is not for cb3 robots.

if so, is there any method to control a robotiq gripper using ROS2 and Moveit?

closed 2025-02-06 by urfeex