UniversalRobots/Universal_Robots_ROS2_Driver #1111: Polyscope 5.15 only works when ran headless:=true

lcbw, 2024-09-24:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

Polyscope 5.15

### How is the ROS driver used.

Headless without using the teach pendant, Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The default mode of the ROS2 driver is to run headless_mode:=true 

External control program made per [these instructions](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/install_urcap_e_series.html) fails to communicate with the UR ROS2 humble driver on our in-house e-series robots on (specifically) Polyscope 5.15. When clicking "play" on the teach pendant program, while the ROS2 driver is running on a user PC, the teach program hangs on a screen saying "starting program". When the ROS2 node is shut down on the PC side, the teach pendant then shows a strange error with the teach pendant program. This has been tested on both a UR5e and a UR10e. 
This error does not appear on Polyscope versions 5.8 or 5.11.2. Other versions of Polyscope have not been tested. 

## Issue details
We are trying to control a UR5e robot on polyscope 5.15 with ROS2 humble. 
With the robot connected to a user PC over ethernet we can successfully: 
ping the robot IP and see the current robot state in RViz. 
If the headless_mode:=false in the macro called at ur_description/urdf/ur_macro.xacro, the external control program must be started manually from the teach pendant. 
When the "play" button is pressed, the program hangs with the following screen if the ROS2 driver is running on the connected PC*:
[image]

The robot does not respond to PC based commands at this time.
When the ROS2 driver node is killed, the teach pendant throws the following error: 
[image]

(*If the ROS2 driver is not running on the connected PC, the teach pendant will throw 'The connection to the remote PC at <Your IP Address> could not be established....')

## Steps to Reproduce
Follow[ these instructions](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/install_urcap_e_series.html) to install externalcontrol-1.0.5.urcap onto the e-series robot on Polyscope 5.15. 
The teach pendant program should look like this when open*: 
[image]

*The warning sign on "installation" is referencing the lack of tool definition:
[image]

Start your UR ROS2 driver on your user PC. 

With the teach pendant in local mode, hit 'play' on the external control teach pendant program. 

## Expected Behavior
I expected the externalcontrol teach pendant program to start successfully, generating the connection required for ROS2 control over the robotic system.

## Actual Behavior
The externalcontrol teach pendant program hangs when called via pressing play on the teach pendant in local mode. The "starting  program" icon appears (see image 'starting_program') and the program hangs, not responding to ROS2 commands. Once the ROS2 driver node is killed, the teach pendant error appears (see image 'program_error').

# Workaround Suggestion
Consistently, setting headless_mode:=true in the macro called at ur_description/urdf/ur_macro.xacro and putting the teach pendant into remote mode (see below) allows for successful system control. 

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-09-24:
I just tried with a 5.15.0, 5.15.1, 5.15.2 and 5.17.2 URSim and could not reproduce the problem.

Does the log tab on the teach pendant give any information? When running correctly, it should look something like this:

[image]

urrsk, 2024-09-25:
@lcbw the issue is that the URCap cannot connect to the ROS driver and receive the script code. 
The External Control URCap should connect to the ROS2 driver when you start or save the program. In this case, the connection times out and the URCap inserts a 'null' instead of the program from the driver.
Therefore, please verify the IP address and port number you are using to connect to the ROS2 driver.

BTW, I can see you are using port 50006 in the picture. The ROS2 driver is by default list on port 50002, unless you have specified something else when you launch the ROS2 driver. I will suggest you try to change it to 50002.

fmauch, 2024-09-25:
As far as I am aware, @lbbw is using a dual arm setup, so changing the ports in one of them seems reasonable. But that could actually be the problem:

According to https://github.com/lcbw/dual_arm_test/blob/main/urdf/env.xacro you are setting 50006 as the `reverse_port` of ur5e, while the `script_sender_port` is set to 50007. That (50007) is the one you should setup in the External Control URCap.

That's also why you don't get the "cannot connect..." popup, since that is a valid port, but just not the one for receiving the script code.

lcbw, 2024-09-25:
@urrsk, I'm trying to follow the format of[ the draft 2 arm tutorial in UR ROS2 Tutorials ](https://github.com/UniversalRobots/Universal_Robots_ROS2_Tutorials/pull/3/files#diff-8fab5c3831a17e05c5a79a461f2b6ad8c1803625044b797e084f9436be2e4ed9).
I can verify I am connected to the correct robot IP & can ping the IP.

@fmauch Thank you for the clarification on the port purposes. Where could I have found documentation regarding this? When I look at [these instructions](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/install_urcap_e_series.html), it says "just don't change the port", which doesn't work in a 2-arm system. I was utilizing the reverse port because it's the only port mentioned and explicitly discussed in [this open documentation PR](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/pull/70/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5). 
[image], but I understand my interpretation was wrong. 
The person who wrote that open documentation PR was *also* having difficulty connecting outside of using headless mode, and may have made the same mistake as I did. 

I have implemented your feedback: 
The settings on the UR10e are set to 172.31.1.137, it can be pinged, and the port is set to 50002. 
The settings on the UR5e are set to 172.31.1.138, it can be pinged, and the port is set to 50007. 
The error is different, it's just the standard "Connection Refused" but I still cannot control either of these robots with headless_mode:=false.

urrsk, 2024-09-26:
@lcbw is should work. And especially when you say that it works in headless mode.
 
Can I ask you to verify the external control URCap using port 50007 by the simple_exterrnal_control_server on the URCap repo https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/tree/master/examples/simple_external_control_server 

Please run the simple server python script from the same machine you are running the ROS2 driver on, to give it the same communication setting as the ROS2 driver.

lcbw, 2024-10-02:
I have gotten it to work ensuring that: 
 The settings on the UR10e are set to 172.31.1.137, it can be pinged, and the port is set to 50002.
 The settings on the UR5e are set to 172.31.1.138, it can be pinged, and the port is set to 50007.
The script_sender_port for the robots are set to the respective port settings from the external control program (50002 for robot 1 and 50007 for robot 2 ). 
The reverse_ip for both robots is set manually to the IP of my computer, 172.31.1.17. 

thank you for your help!

closed 2024-10-02 by lcbw