UniversalRobots/Universal_Robots_ROS2_Driver #1116: C271A1 Low level real-time thread issue on dual arm setup

kfryhilderbrand, 2024-10-01:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with realtime patch

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

Polyscope 5.15

### How is the ROS driver used.

Headless without using the teach pendant, Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Dual arm setup being controlled from 1 controller manager, trajectory being sent to one robot that fails to execute the trajectory due to a real-time thread error.

## Issue details
Running a setup with two UR robots (UR5e and UR10e). The UR10e is running an admittance controller while the UR5e is running a joint trajectory controller (1 controller manager). We are sending a trajectory to the UR5e, but it throws a "Low level real-time thread" error and does not execute the trajectory. We tried jogging the robot to a consistent starting pose (that the trajectory also starts from), but that did not solve the issue.

## Expected Behavior
Admittance controller running on UR10e, robot will not move if COM calculations correct. Joint trajectory controller running on UR5e, trajectory being sent and executed on robot.

## Actual Behavior
Admittance controller on UR10e ran without issue, trajectory did not run on the UR5e

# Workaround Suggestion
Suggested including a Wait or a sync(), but including these within our trajectory executor did not solve the issue.

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-10-09:
Hi, does this only happen in the dual-arm setup? It would be good to get this to a reproducible minimal configuration so we can start debugging this.

lcbw, 2024-10-14:
Hello @fmauch no - this problem also occurs when the UR5e is operated alone, but not when the UR10e is operated alone. 
Please try running the following command with the recently updated dual arm test repository main branch. 
ros2 launch dual_arm_test dual_arm_test.launch.py single_arm:=true use_fake_hardware:=false fake_sensor_commands:=false

The updates to this branch include fixes to the single_arm launch parameter, which should generate a simplified robot description, and a hardcoded update to the controllers.yaml file to only reference robot1.

urfeex, 2025-03-24:
Closing this due to inactivity.

closed 2025-03-24 by urfeex