UniversalRobots/Universal_Robots_ROS2_Driver #1129: Scaled_joint_trajectory_controller does not respond to speed change on teach pendant

Adrian2702, 2024-10-08:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.13.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Controlling a UR10e robot with the URCap “External-Control” from an external PC. Waypoints are sent from the PC to the real robot, which it also follows, but the movement is aborted due to a time-out error as soon as the speed slider is not at 100%

## Issue details
We are trying to control a UR10e robot with Polyscope 5.13 via ROS2 Humble. The robot is connected to an external PC via Ethernet. The URCap “External-Control” is installed on the real robot. When I press the play button, I get the following output on the screen:
```
[ros2_control_node-1] [INFO] [1728380483.965475739] [UR_Client_Library:]: Robot requested program
[ros2_control_node-1] [INFO] [1728380483.965620055] [UR_Client_Library:]: Sent program to robot
[ros2_control_node-1] [INFO] [1728380484.260967978] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
```
The Scaled_joint_trajectory_controller is active and receives waypoints via a C++ script. Everything works well and the robot follows the path. However, this is only possible if the speed controller is set to 100%. If I slow it down, I get the following error:
```
MoveGroupInterface::execute() failed or timeout reached
```
The robot moves to the first points of the path and then aborts the movement with the MoveGroup Error.

## Steps to Reproduce
Start the Launch-Files``` ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.1.102 launch_rviz:=false ``` and ```ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10e```

## Expected Behavior
The robot will slow down his movements, when the speed slider on the teach pendant is scaled down but don't aborted the movement.

## Actual Behavior
The robot moves to the first few points and then aborted the movement.

# Workaround Suggestion
Maybe I misunderstand the scaled_joint_trajectory_controller, but I thought this was a special controller for the UR series that takes into account the current value of the speed controller. I can also read the corresponding topic (```ros2 topic echo /speed_scaling_state_broadcaster/speed_scaling```) and get the same values as on the teach pendant. What other configurations need to be made on the controller? I have checked the ```ur_controllers.yaml``` and ```moveit_controllers.yaml```, but I have no idea where the problem is coming from.

Thanks in advance for any helpful tips!

### Relevant log output

```shell
cobot@MBS-Cobot:~$ ros2 launch ur_robot_driver ur10e_control_moveit.launch.py
[INFO] [launch]: All log files can be found below /home/cobot/.ros/log/2024-10-08-14-38-46-430353-MBS-Cobot-94879
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/home/cobot/lerosh_ws/install/ur_moveit_config/share/ur_moveit_config`. -- using config/ur.urdf
WARNING:root:Cannot infer SRDF from `/home/cobot/lerosh_ws/install/ur_moveit_config/share/ur_moveit_config`. -- using config/ur.srdf
WARNING:root:"File /home/cobot/lerosh_ws/install/ur_moveit_config/share/ur_moveit_config/config/ur.urdf doesn't exist"
WARNING:root:The robot description will be loaded from /robot_description topic 
[... 197 lines cut ...]
[ros2_control_node-1] [INFO] [1728391130.093682455] [scaled_joint_trajectory_controller]: Using 'splines' interpolation method.
[ros2_control_node-1] [INFO] [1728391130.098170640] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-8] [INFO] [1728391130.129911743] [spawner_joint_state_broadcaster]: Configured and activated io_and_status_controller
[spawner-7] [INFO] [1728391130.133290569] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[ros2_control_node-1] [INFO] [1728391130.134756124] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[INFO] [spawner-9]: process has finished cleanly [pid 94891]
[ros2_control_node-1] [INFO] [1728391130.151169462] [speed_scaling_state_broadcaster]: Loading UR SpeedScalingStateBroadcaster with tf_prefix: 
[spawner-8] [INFO] [1728391130.156187964] [spawner_joint_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1728391130.158451223] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1728391130.158620357] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[spawner-8] [INFO] [1728391130.169530409] [spawner_joint_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1728391130.175772237] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-8] [INFO] [1728391130.199331945] [spawner_joint_state_broadcaster]: Loaded force_torque_sensor_broadcaster
[ros2_control_node-1] [INFO] [1728391130.201427300] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[spawner-8] [INFO] [1728391130.213443778] [spawner_joint_state_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[INFO] [spawner-7]: process has finished cleanly [pid 94889]
[INFO] [spawner-8]: process has finished cleanly [pid 94890]
[ros2_control_node-1] [INFO] [1728391161.605165820] [UR_Client_Library:]: Robot requested program
[ros2_control_node-1] [INFO] [1728391161.605328280] [UR_Client_Library:]: Sent program to robot
[ros2_control_node-1] [INFO] [1728391161.884950396] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-10-08:
I assume that this is the problem:

```
MoveGroupInterface::execute() failed or timeout reached
```

If that's the case, it's probably not that the controller doesn't use the scaling from the teach pendant, but the MoveIt trajectory execution monitoring. Could you try #1132 fixes that problem for you?

Adrian2702, 2024-10-09:
Thank you very much for your quick reply. I have added the following lines in the ```moveit_controllers.yaml``` file from the package ```ur_moveit_config``` as you described in #1132:

```
trajectory_execution:
  allowed_execution_duration_scaling: 1.2
  allowed_goal_duration_margin: 0.5
  allowed_start_tolerance: 0.01
  execution_duration_monitoring: false # Not much use when using the scaled JTC
```

I have tested it with the UR10e robot. It works like a charm. Both under humble and jazzy.

PR #1133 (2024-10-10): [moveit] Disable execution_duration_monitoring by default
This should close #1129
closed 2024-10-09 by Adrian2702