UniversalRobots/Universal_Robots_ROS2_Driver #1141: UR10 Robot Not Responding to Commands Using ROS2 Humble

johnprada94, 2024-10-11:
Hello Universal Robots Team/Community,

I am currently facing an issue where my UR10 robot is not moving as expected when using ROS2 Humble with the `forward_velocity_controller`. Below is a detailed description of the problem, setup, and troubleshooting steps I have taken.

### **Setup Details:**

- Robot Model: UR10
- ROS2 Version: Humble
- UR ROS2 Driver Package: ur_robot_driver
- Operating System: Ubuntu 22.04
- URCap installed on robot: Yes (Verified working)

### **Steps Performed:**

1. Successfully connected the UR10 robot to my ROS2 system using the command:
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10 robot_ip:=192.168.0.100`

> 
> [INFO] [launch]: All log files can be found below /home/ur10/.ros/log/2024-10-11-10-53-16-247202-ur10-4964
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [dashboard_client-2]: process started with pid [4968]
> [INFO] [controller_stopper_node-3]: process started with pid [4970]
> [INFO] [ur_ros2_control_node-1]: process started with pid [4966]
> [INFO] [urscript_interface-4]: process started with pid [4972]
> [INFO] [robot_state_publisher-5]: process started with pid [4974]
> [INFO] [rviz2-6]: process started with pid [4976]
> [INFO] [spawner-7]: process started with pid [4978]
> [INFO] [spawner-8]: process started with pid [4980]
> [INFO] [spawner-9]: process started with pid [4982]
> [dashboard_client-2] [INFO] [1728611596.687380269] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
> [dashboard_client-2] 
> [controller_stopper_node-3] [INFO] [1728611596.688876008] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
> [rviz2-6] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
> [robot_state_publisher-5] [INFO] [1728611596.697821643] [robot_state_publisher]: got segment base
> [robot_state_publisher-5] [INFO] [1728611596.697928392] [robot_state_publisher]: got segment base_link
> [robot_state_publisher-5] [INFO] [1728611596.697942278] [robot_state_publisher]: got segment base_link_inertia
> [robot_state_publisher-5] [INFO] [1728611596.697949649] [robot_state_publisher]: got segment flange
> [robot_state_publisher-5] [INFO] [1728611596.697956204] [robot_state_publisher]: got segment forearm_link
> [robot_state_publisher-5] [INFO] [1728611596.697962814] [robot_state_publisher]: got segment ft_frame
> [robot_state_publisher-5] [INFO] [1728611596.697969474] [robot_state_publisher]: got segment shoulder_link
> [robot_state_publisher-5] [INFO] [1728611596.697975955] [robot_state_publisher]: got segment tool0
> [robot_state_publisher-5] [INFO] [1728611596.697982446] [robot_state_publisher]: got segment upper_arm_link
> [robot_state_publisher-5] [INFO] [1728611596.698004825] [robot_state_publisher]: got segment world
> [robot_state_publisher-5] [INFO] [1728611596.698051793] [robot_state_publisher]: got segment wrist_1_link
> [robot_state_publisher-5] [INFO] [1728611596.698059689] [robot_state_publisher]: got segment wrist_2_link
> [robot_state_publisher-5] [INFO] [1728611596.698065915] [robot_state_publisher]: got segment wrist_3_link
> [ur_ros2_control_node-1] [WARN] [1728611596.700538345] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
> [ur_ros2_control_node-1] text not specified in the tf_prefix tag
> [ur_ros2_control_node-1] [INFO] [1728611596.700965663] [resource_manager]: Loading hardware 'ur10' 
> [ur_ros2_control_node-1] [INFO] [1728611596.704154838] [resource_manager]: Initialize hardware 'ur10' 
> [ur_ros2_control_node-1] [INFO] [1728611596.704221002] [resource_manager]: Successful initialization of hardware 'ur10'
> [ur_ros2_control_node-1] [INFO] [1728611596.704360965] [resource_manager]: 'configure' hardware 'ur10' 
> [ur_ros2_control_node-1] [INFO] [1728611596.704372179] [URPositionHardwareInterface]: Starting ...please wait...
> [ur_ros2_control_node-1] [INFO] [1728611596.704382914] [URPositionHardwareInterface]: Initializing driver...
> [ur_ros2_control_node-1] [WARN] [1728611596.704986498] [UR_Client_Library:]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
> [ur_ros2_control_node-1] [INFO] [1728611596.814544266] [UR_Client_Library:]: Negotiated RTDE protocol version to 2.
> [ur_ros2_control_node-1] [INFO] [1728611596.814806592] [UR_Client_Library:]: Setting up RTDE communication with frequency 125.000000
> [spawner-7] [INFO] [1728611596.935520567] [spawner_scaled_joint_trajectory_controller]: waiting for service /controller_manager/list_controllers to become available...
> [spawner-9] [INFO] [1728611596.935605627] [spawner_forward_position_controller]: waiting for service /controller_manager/list_controllers to become available...
> [spawner-8] [INFO] [1728611596.945182652] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
> [rviz2-6] [INFO] [1728611597.187779596] [rviz2]: Stereo is NOT SUPPORTED
> [rviz2-6] [INFO] [1728611597.187974736] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
> [rviz2-6] [INFO] [1728611597.202802918] [rviz2]: Stereo is NOT SUPPORTED
> [ur_ros2_control_node-1] [INFO] [1728611597.890879505] [URPositionHardwareInterface]: Calibration checksum: 'calib_17227329492635474227'.
> [ur_ros2_control_node-1] [ERROR] [1728611598.966232571] [URPositionHardwareInterface]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_calibration/README.md] for details.
> [ur_ros2_control_node-1] [INFO] [1728611598.966394492] [URPositionHardwareInterface]: System successfully started!
> [ur_ros2_control_node-1] [INFO] [1728611598.966446370] [resource_manager]: Successful 'configure' of hardware 'ur10'
> [ur_ros2_control_node-1] [INFO] [1728611598.966580039] [resource_manager]: 'activate' hardware 'ur10' 
> [ur_ros2_control_node-1] [INFO] [1728611598.966603135] [URPositionHardwareInterface]: Activating HW interface
> [ur_ros2_control_node-1] [INFO] [1728611598.966621277] [resource_manager]: Successful 'activate' of hardware 'ur10'
[... 142 lines cut ...]

**ROS2 Controller Status:**

I checked the controller status using:

`ros2 control list_controllers`

> [INFO] [1728624143.633446703] [_ros2cli_9633]: waiting for service /controller_manager/list_controllers to become available...
> scaled_joint_trajectory_controller ur_controllers/ScaledJointTrajectoryController                inactive
> forward_position_controller        position_controllers/JointGroupPositionController             active  
> joint_state_broadcaster            joint_state_broadcaster/JointStateBroadcaster                 active  
> io_and_status_controller           ur_controllers/GPIOController                                 active  
> speed_scaling_state_broadcaster    ur_controllers/SpeedScalingStateBroadcaster                   active  
> force_torque_sensor_broadcaster    force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster  active  

I attempted to activate the scaled_joint_trajectory_controller using:
`ros2 control switch_controllers --start scaled_joint_trajectory_controller`

But even after activation, **the robot does not move.**

Is there anything I am missing in activating and utilizing the controllers in ROS2?

Any help or insights would be greatly appreciated!

Thank you for your time.

vincentkeino, 2024-11-01:
i have same problem too，could u tell me how to sove this problem？thx！

johnprada94, 2024-11-08:
> i have same problem too，could u tell me how to sove this problem？thx！

In my case, i found out that the teach pendant was having two URCaps activated. I only left the External Control active and I delete all others

urfeex, 2025-02-06:
As this seems to be resolved for the author, I will close this issue.

closed 2025-02-06 by urfeex