UniversalRobots/Universal_Robots_ROS2_Driver #1143: Issue name

joshikaustubh, 2024-10-11:
### Affected ROS2 Driver version(s)

ur_controllers

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

URSim in a virtual machine

### Robot SW / URSim version(s)

Gazebo

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary

I am trying to build the drivers from source fr certain modifications

## Issue details

I am getting an error when running `colcon build` after sourcing the package for the `humble` branch

## Steps to Reproduce

1. `git clone -b humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git`
2. Build the workspace
3. Error when building ur_controllers

## Expected Behavior

I expected a successful build

## Actual Behavior

[image]

# Workaround Suggestion

I noticed this is happening since last 2-3 days. The version I had from last week was working totally fine. I don't know if that's the issue.

### Accept Public visibility

- [X] I agree to make this context public

kukakukaka, 2024-10-14:
I've also run into this bug, is there a workaround for now?

gunwoo0623, 2024-10-14:
I also encountered the exactly the same issue with this for 1 week. However, one thing i discovered is that my colleague could do 'colcon build' successfully without problems but, mine cannot. Please someone leave the comments for this issue.

fmauch, 2024-10-14:
We are currently in the process of adding new features. Is there a reason why you would like to build from source? Otherwise I would recommend using the available apt packages as suggested in the [installation documentation](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/installation.html).

In case you really want to build from source, you'll also have to build the upstream packages from source as mentioned in point 6 of that docs.

adam-malyshev, 2024-10-15:
I get the same issue when running `colcon build` when using the available apt packages through `rosdep install --ignore-src --from-paths src -y`. Same for me it worked a couple weeks ago, but now is not because of the latest commit.

fmauch, 2024-10-15:
This is **normal and** expected until the next sync. If you want to build the driver from source, use an all-source build where you also build ur_msgs from source as mentioned above.

fmauch, 2024-10-16:
On Thursday there will be a Humble sync, with which this should go away.

fmauch, 2024-10-18:
Humble binary is building again.

closed 2024-10-18 by fmauch