UniversalRobots/Universal_Robots_ROS2_Driver #1157: Driver stability issue

AleTarsi, 2024-10-17:
### Affected ROS2 Driver version(s)

2.2.10

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.9

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I sporadically lose the connection with my UR5 robot, when using the ROS2 driver, but never when `use_fake_hardware` is set to `True`. The problem takes place at the driver startup and brings two possible different faults: 

1. The `joint states` topic is not published and the robot is not visible in Rviz, even though the `robot_description` topic is published together with the `robot_state_publisher` node. All my `ros2 controllers`, also are not spawned because they encounter a timeout after 30 seconds. This causes the robot to be not visible in Rviz and the easy solution to solve the problem is just restarting everything, which solves the problem almost always.  
2. The driver is partially loaded except for the `ros2 controller` I need to move the robot: `forward position controller`. This problem can be solved by either restarting all my nodes or asking again to load and activate the controller.

## Issue details
I haven’t found a solution to prevent this behavior, which is annoying, especially in the first case. I tried inserting a timer in the launch file to spread the node creation overhead:

I start by launching the core nodes for the robot communication which are:

```bash
nodes_to_start = [
        control_node,
        ur_control_node,
        dashboard_client_node,
        tool_communication_node,
        controller_stopper_node,
        urscript_interface,
        robot_state_publisher_node,
    ] + controller_spawners
```

After 10.0 seconds I launch also MoveIt and other service nodes I need for my application in other launch files. The controller spawner timeout is currently set to 20.0 seconds.

This is the code of my launch file for the driver:

<details>
<summary> Launch file UR5 </summary>

```py

...
from launch_ros.actions import Node
from launch_ros.parameter_descriptions import ParameterFile
from launch_ros.substitutions import FindPackageShare
from launch import LaunchDescription
from launch.conditions import IfCondition, UnlessCondition
from launch.substitutions import Command, FindExecutable, LaunchConfiguration, PathJoinSubstitution
from ament_index_python.packages import get_package_share_directory
[... 400 lines cut ...]
<summary> Case robot doesn’t spawn in Rviz </summary>

[image]

</details>

<details>
<summary> Case robot doesn’t move  </summary>

[image]

</details>

## Expected Behavior
The robot connection spawns correctly every time the system is launched. 

## Actual Behavior
Sporadically the robot doesn’t move or doesn’t show up due to nodes timeout and missing communication with the real robot.

# Workaround Suggestion
Restart the application or restart missing nodes.

### Accept Public visibility

- [X] I agree to make this context public

AleTarsi, 2024-10-17:
<details>
<summary> Log output in case the robot doesn’t move</summary>

```sh
[INFO] [launch]: All log files can be found below /home/icub/.ros/log/2024-10-16-17-47-44-934091-iiticublap158u-1243222
[INFO] [launch]: Default logging verbosity is set to INFO
Using calibrated kinematics file
[INFO] [dashboard_client-2]: process started with pid [1243234]
[INFO] [ros2-1]: process started with pid [1243232]
[INFO] [urscript_interface-3]: process started with pid [1243236]
[INFO] [robot_state_publisher-4]: process started with pid [1243238]
[INFO] [controller_stopper_node-5]: process started with pid [1243240]
[dashboard_client-2] [INFO] [1729093666.527589984] [UR_Client_Library:]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[robot_state_publisher-4] [INFO] [1729093666.535651322] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1729093666.535756056] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1729093666.535773999] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1729093666.535785677] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1729093666.535794842] [robot_state_publisher]: got segment camera_link_gazebo
[robot_state_publisher-4] [INFO] [1729093666.535805006] [robot_state_publisher]: got segment camera_link_optical
[robot_state_publisher-4] [INFO] [1729093666.535821019] [robot_state_publisher]: got segment end_effector
[robot_state_publisher-4] [INFO] [1729093666.535837124] [robot_state_publisher]: got segment finger_1
[robot_state_publisher-4] [INFO] [1729093666.535849181] [robot_state_publisher]: got segment finger_2
[robot_state_publisher-4] [INFO] [1729093666.535862695] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1729093666.535876292] [robot_state_publisher]: got segment flange_link
[robot_state_publisher-4] [INFO] [1729093666.535890702] [robot_state_publisher]: got segment flange_to_camera
[robot_state_publisher-4] [INFO] [1729093666.535903780] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1729093666.535918357] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1729093666.535929662] [robot_state_publisher]: got segment rack_1
[robot_state_publisher-4] [INFO] [1729093666.535943580] [robot_state_publisher]: got segment rack_2
[robot_state_publisher-4] [INFO] [1729093666.535961318] [robot_state_publisher]: got segment root_link
[robot_state_publisher-4] [INFO] [1729093666.535977880] [robot_state_publisher]: got segment shaft
[robot_state_publisher-4] [INFO] [1729093666.535994069] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1729093666.536008984] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1729093666.536020632] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1729093666.536032215] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1729093666.536042521] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1729093666.536051471] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1729093666.536059940] [robot_state_publisher]: got segment wrist_3_link
[controller_stopper_node-5] [INFO] [1729093666.546082340] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ros2-1] stdin is not a terminal device. Keyboard handling disabled.[INFO] [1729093666.738990768] [rosbag2_recorder]: Press SPACE for pausing/resuming
[ros2-1] [INFO] [1729093666.740628997] [rosbag2_storage]: Opened database '/media/icub/data/local_data/jl2_workspace/bag/bag_0.db3' for READ_WRITE.
[ros2-1] [INFO] [1729093666.741116313] [rosbag2_recorder]: Listening for topics...
[ros2-1] [INFO] [1729093666.741125442] [rosbag2_recorder]: Event publisher thread: Starting
[ros2-1] [INFO] [1729093666.743868077] [rosbag2_recorder]: Subscribed to topic '/rosout'
[ros2-1] [INFO] [1729093666.745802330] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[ros2-1] [INFO] [1729093666.746408648] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[ros2-1] [INFO] [1729093666.746507321] [rosbag2_recorder]: Recording...
[ros2-1] [INFO] [1729093666.849009411] [rosbag2_recorder]: Subscribed to topic '/robot_description'
[ros2-1] [INFO] [1729093666.850216449] [rosbag2_recorder]: Subscribed to topic '/tf_static'
[ros2-1] [INFO] [1729093666.850903909] [rosbag2_recorder]: Subscribed to topic '/tf'
[INFO] [ur_ros2_control_node-6]: process started with pid [1243299]
[ur_ros2_control_node-6] [WARN] [1729093669.827847064] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ur_ros2_control_node-6] text not specified in the tf_prefix tag
[ur_ros2_control_node-6] [INFO] [1729093669.828869866] [resource_manager]: Loading hardware 'ur' 
[ur_ros2_control_node-6] [INFO] [1729093669.830817040] [resource_manager]: Initialize hardware 'ur' 
[ur_ros2_control_node-6] [INFO] [1729093669.830862870] [resource_manager]: Successful initialization of hardware 'ur'
[ur_ros2_control_node-6] [INFO] [1729093669.831009246] [resource_manager]: 'configure' hardware 'ur' 
[... cut ...]
[ros2-1] [INFO] [1729073874.707584269] [rosbag2_recorder]: Subscribed to topic '/io_and_status_controller/robot_mode'
[ros2-1] [INFO] [1729073874.709863215] [rosbag2_recorder]: Subscribed to topic '/io_and_status_controller/io_states'
[INFO] [spawner-19]: process has finished cleanly [pid 1120503]
[INFO] [spawner-17]: process has finished cleanly [pid 1120499]
[INFO] [spawner-16]: process has finished cleanly [pid 1120497]
[INFO] [spawner-18]: process has finished cleanly [pid 1120501]
```

</details>

fmauch, 2024-10-18:
This looks like you are running into the problems we solved in https://github.com/ros-controls/ros2_control/pull/1680 and https://github.com/ros-controls/ros2_control/pull/1686.

Just to make sure: Do you have a up-to-date ros2_control installation?

To improve things, you could try to reduce the number of spawners, since the spawner now accepts a list of controllers.

In our launchfile we use two spawners for the fixed lists instead of the 6 that you are currently using (as we did in the past, as well):

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/255355d7876f1cec996f957107c548f9848cf7ed/ur_robot_driver/launch/ur_control.launch.py#L321-L347

Depending on which controller is missing, behavior will be different, hence the different faults you encounter.

AleTarsi, 2024-10-21:
> Just to make sure: Do you have a up-to-date ros2_control installation?

I am using a conda environment for the dependencies, for this reason I may not be able to always have the latest version of the all the packages. As you can see I rely on `robostack` for the my project dependencies.

Here is my mamba list with the versions of all my ros packages: 

<details>
<summary> Mamba List </summary>

```bash
# Name                    Version                   Build  Channel

ros-humble-ackermann-msgs 2.0.2           py311hb335429_6    robostack-staging
ros-humble-ackermann-steering-controller 2.32.0          py311hb335429_6    robostack-staging
ros-humble-action-msgs    1.2.1           py311hb335429_6    robostack-staging
ros-humble-action-tutorials-cpp 0.20.3          py311hb335429_6    robostack-staging
ros-humble-action-tutorials-interfaces 0.20.3          py311hb335429_6    robostack-staging
ros-humble-action-tutorials-py 0.20.3          py311hb335429_6    robostack-staging
ros-humble-actionlib-msgs 4.2.3           py311hb335429_6    robostack-staging
ros-humble-admittance-controller 2.32.0          py311hb335429_6    robostack-staging
ros-humble-ament-cmake    1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-auto 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-copyright 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-core 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-cppcheck 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-cpplint 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-definitions 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-dependencies 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-include-directories 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-interfaces 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-libraries 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-link-flags 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-export-targets 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-flake8 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-gen-version-h 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-gmock 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-gtest 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-include-directories 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-libraries 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-lint-cmake 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-pep257 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-pytest 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-python 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-ros 0.10.0          py311hb335429_6    robostack-staging
ros-humble-ament-cmake-target-dependencies 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-test 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-uncrustify 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cmake-version 1.3.7           py311hb335429_6    robostack-staging
ros-humble-ament-cmake-xmllint 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-copyright 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cppcheck 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-cpplint  0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-flake8   0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-index-cpp 1.4.0           py311hb335429_6    robostack-staging
ros-humble-ament-index-python 1.4.0           py311hb335429_6    robostack-staging
ros-humble-ament-lint     0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-lint-auto 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-lint-cmake 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-lint-common 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-package  0.14.0          py311hb335429_6    robostack-staging
ros-humble-ament-pep257   0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-uncrustify 0.12.10         py311hb335429_6    robostack-staging
ros-humble-ament-xmllint  0.12.10         py311hb335429_6    robostack-staging
ros-humble-angles         1.15.0          py311hb335429_6    robostack-staging
ros-humble-backward-ros   1.0.2           py311h4ff203a_6    robostack-staging
ros-humble-bicycle-steering-controller 2.32.0          py311hb335429_6    robostack-staging
ros-humble-builtin-interfaces 1.2.1           py311hb335429_6    robostack-staging
ros-humble-camera-calibration-parsers 3.1.8           py311hb335429_6    robostack-staging
ros-humble-camera-info-manager 3.1.8           py311hb335429_6    robostack-staging
ros-humble-class-loader   2.2.0           py311hb303436_6    robostack-staging
[... 320 lines cut ...]
ros-humble-urdf           2.6.0           py311hb335429_6    robostack-staging
[... cut ...]
ros2-distro-mutex         0.5.0                    humble    robostack-staging
rosdistro                 0.9.0           py311h38be061_1    conda-forge
rospkg                    1.5.1              pyhd8ed1ab_0    conda-forge

```

</details>

As you can see `ros-humble-ros2-control == 2.38.0` , the latest release should be `2.43.1`

> To improve things, you could try to reduce the number of spawners, since the spawner now accepts a list of controllers.

I have implemented your suggestions, I will now perform some tests and get back to you.

AleTarsi, 2024-10-28:
Your suggestion helped, at least, reducing the frequency of the problem. So far I haven't encountered any troubles related to the controllers not spawning

urfeex, 2025-02-06:
I'm going to close this since I didn't hear anything back and it seems to work for you. Feel free to comment / reopen if the issue still persists.

closed 2025-02-06 by urfeex