UniversalRobots/Universal_Robots_ROS2_Driver #1158: ur_msgs is not present in Universal_Robots_ROS2_Driver and the humble ROS version of this package does not compile on ROS2

ctorresu, 2024-10-19:
### Feature summary

**Introduction**
I'm doing the ur_robot_driver installation tutorial and when compiling I get an error related to the ut_msgs package which is not included in Universal_Robots_ROS2_Driver. To fix it, I install the humble version of the package present in this github repository: https://github.com/ros-industrial/ur_msgs/tree/humble-devel
However, when building the package I get an error related to the catkin_make constructor which is specific to ROS1 and not ROS2.

**Longer Description**
I am following the instructions to install the UR driver in Ubuntu 22.04, ROS2 humble in the official page of Universal Robots:
https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html

When compiling the package I get this error:
rkspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/ur_configuration_controller.hpp:52:10: fatal error: ur_msgs/srv/get_robot_software_version.hpp: No such file or directory
   52 | #include "ur_msgs/srv/get_robot_software_version.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:118: CMakeFiles/ur_controllers.dir/src/ur_configuration_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:252: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [27.3s, exited with code 2]
                                     
Summary: 3 packages finished [1min 2s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  4 packages not processed

I notice that the error is related because this package is not in Universal_Robots_ROS2_Driver and to solve the problem I install the version of ur_msgs for humble that is found in the following link:
https://github.com/ros-industrial/ur_msgs/tree/humble-devel

However, after recompiling I receive the following error:
--- stderr: ur_msgs                                                                                 
CMake Error at CMakeLists.txt:7 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

I don't understand why if I'm downloading the package for the humble ROS version I'm having problems building for catkin which are from ROS1. Please, I need help to be able to install the driver and connect Moveit2 with my UR10 robot.

fmauch, 2024-10-20:
From your output it seems that there are two possible issues:

1. You already had `ur_msgs` installed, but your system is not up-to-date. We recently added that service to ur_msgs and that got synchronized to the main repository last thursday. In that case, a simple `sudo apt update && sudo apt dist-upgrade` should be sufficient to get the most recent `ur_msgs` version installed.
2. It seems that while you cloned the correct ur_msgs repo, you did not checkout the `humble-devel` branch after cloning but you are trying to build from the default branch which is currently an older ROS 1 version.

As the first step should make installing ur_msgs from source unnecessary I would recommend deleting the source-ur_msgs package before building again.

urfeex, 2024-11-28:
Closing due to no updates with a proposed solution. If the error still persists, feel free to comment / reopen this issue.

closed 2024-11-28 by urfeex