UniversalRobots/Universal_Robots_ROS2_Driver #1174: ROS2 build error - No such file or directory

junhui324, 2024-11-01:
### Affected ROS2 Driver version(s)

rolling (main branch)

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

URSim in a virtual machine

### Robot SW / URSim version(s)

5.12.6

### How is the ROS driver used.

Others

### Issue details

# Summary
I use ROS2 rolling and git clone main branch

I follow this page. (https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html)

But, I had an error at this step. -> colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

## Issue details
```
cd $COLCON_WS
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.${ROS_DISTRO}.repos
rosdep update
rosdep install --ignore-src --from-paths src -y
```
As a result of proceeding to the above step, the ur_msgs folder was created.
And when I proceeded with the colcon build again, the following error occurred.

```
Starting >>> ur_moveit_config
--- stderr: ur_controllers                                                
In file included from /home/jjh/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp:41:
/home/jjh/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/ur_configuration_controller.hpp:52:10: fatal error: ur_msgs/srv/get_robot_software_version.hpp: No such file or directory
   52 | #include "ur_msgs/srv/get_robot_software_version.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:118: CMakeFiles/ur_controllers.dir/src/ur_configuration_controller.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/jjh/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/src/gpio_controller.cpp:38:
/home/jjh/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/gpio_controller.hpp:54:10: fatal error: ur_msgs/srv/set_analog_output.hpp: No such file or directory
   54 | #include "ur_msgs/srv/set_analog_output.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:104: CMakeFiles/ur_controllers.dir/src/gpio_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:252: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [1.80s, exited with code 2]
Aborted  <<< ur_moveit_config [0.10s]

Summary: 22 packages finished [24.9s]
  1 package failed: ur_controllers
  1 package aborted: ur_moveit_config
  2 packages had stderr output: controller_manager ur_controllers
  22 packages not processed
```

The inside of the ur_msgs/srv folder is shown in the picture below.
[image]

I'm wondering if I made a mistake somewhere or if I need to download the .hpp files separately from somewhere else.

Sorry for not being good at English. Any help would be appreciated.🥲

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2024-11-28:
Maybe the `--skip-existing` flag is actually a bad advice here. I presume that your copy of ur_msgs is simply too old, probably on the wrong branch. It should be on `humble-devel`

urfeex, 2025-01-15:
Closing this due to a proposed solution without any update. Feel free to reopen if the problem still persists.

closed 2025-01-15 by urfeex