UniversalRobots/Universal_Robots_ROS2_Driver #1179: No kinematics plugins defined. Fill and load kinematics.yaml!

sarmadahmad8, 2024-11-07:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

ur5

### How is the ROS driver used.

Others

### Issue details

When i run the moveit support package for this driver through:
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5 launch_rviz:=true
i get this warning 
[WARN] [1731009987.767804840] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
However this is not a problem when controlling the robot through the rviz gui using moveit.
The problem arises when im trying to run my own move_group node in parallel with the ur_moveit_config package. i get this error when trying to use Inverse kinematics for setting joint value targets from poses in cartesian space.
[ERROR] [1731009989.969819875] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'ur_manipulator'
I used the move group interface tutorial to build my custom node. 
These are the changes i made to the ur_moveit.launch.py:

i added this code to the launch_setup in ur_moveit.launch.py

    end_effector_node_2 = Node(
        package='moving_ur5', 
        executable='move_group_reciever', 
        output='screen',
        parameters=[
            robot_description,
            robot_description_semantic,
            publish_robot_description_semantic,
            robot_description_kinematics,
            robot_description_planning,
            ompl_planning_pipeline_config,
            trajectory_execution,
            moveit_controllers,
            planning_scene_monitor_parameters,
            {"use_sim_time": use_sim_time},
            warehouse_ros_config,
        ],

Here is a snippet of my custom move_group_reciever node:

  int main(int argc, char** argv)
  {
      rclcpp::init(argc, argv);
      rclcpp::NodeOptions node_options;
      node_options.automatically_declare_parameters_from_overrides(true);
      auto move_group_reciever = std::make_shared<MoveGroupReceiver>();
  
      // Single-threaded executor for the robot state monitor
[... 210 lines cut ...]
[INFO] [1731009988.940841671] [move_group_interface]: Plan and Execute request accepted
[INFO] [1731009988.946121148] [move_group_interface]: Plan and Execute request complete!
[INFO] [1731009989.947126059] [move_group_demo]: Planning frame: world
[INFO] [1731009989.947294813] [move_group_interface]: MoveGroup action client/server ready
[INFO] [1731009989.947716380] [move_group_interface]: Planning request accepted
[INFO] [1731009989.954573410] [move_group_interface]: Planning request aborted
[ERROR] [1731009989.954851102] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
[INFO] [1731009989.954882202] [move_group_demo]: Visualizing plan (pose goal) FAILED
[INFO] [1731009989.955369545] [move_group_interface]: Plan and Execute request accepted
[INFO] [1731009989.962616231] [move_group_interface]: Plan and Execute request aborted
[ERROR] [1731009989.963635134] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached
[ERROR] [1731009989.969819875] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'ur_manipulator'
[INFO] [1731009989.969840915] [move_group_demo]: Planning frame: world
[INFO] [1731009989.969941488] [move_group_interface]: MoveGroup action client/server ready
[INFO] [1731009989.970181568] [move_group_interface]: Planning request accepted
[INFO] [1731009990.006926647] [move_group_interface]: Planning request complete!
[INFO] [1731009990.007973022] [move_group_interface]: time taken to generate plan: 0.0234684 seconds
[INFO] [1731009990.008015313] [move_group_demo]: Visualizing plan (pose goal) 
[INFO] [1731009990.008534568] [move_group_interface]: Plan and Execute request accepted
[INFO] [1731009990.010111760] [move_group_interface]: Plan and Execute request complete!
```

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2024-11-28:
`No kinematics plugins defined. Fill and load kinematics.yaml!` in the standard configuration often points to a problem with locale. What is your output for the `locale` command?

If you are using a locale (`LC_NUMERIC` specifically) that uses "," as a decimal separator, things break. As a workaround you can try starting things using 

```
LC_ALL=C ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5
```

I hope that helps.

urfeex, 2025-02-06:
I'll be closing this due to a proposed solution with no update. Feel free to comment / reopen if the issue still persists.

closed 2025-02-06 by urfeex