UniversalRobots/Universal_Robots_ROS2_Driver #1188: robot mode use the same output for normal/running/pause

bodkal, 2024-11-21:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in linux

### Robot SW / URSim version(s)

ros2 run ur_client_library start_ursim.sh -m ur10e

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary

When I read the topic /io_and_status_controller/robot_mode using the command:

`ros2 topic echo /io_and_status_controller/robot_mode
`
I  get the same output for all states: normal, running, and pause.:

`mode: 7
`

## Issue details
I am trying to determine the appropriate time to use the service /io_and_status_controller/resend_robot_program so that I can send trajectory commands to the driver using MoveIt. I am only able to successfully send these commands when the TechPad displays the status as running.

Additionally, when I invoke:

`ros2 service call /io_and_status_controller/resend_robot_program std_srvs/srv/Trigger "{}"
`
It does not trigger any observable change in the ROS2 topic /io_and_status_controller/robot_mode.

How can I accurately determine the actual robot state? Is there another method or workaround to differentiate these states?

## Steps to Reproduce
start sim adn driver.

run 

in terminal 1 "ros2 topic echo /io_and_status_controller/robot_mode
in terminal 2 "ros2 service call /dashboard_client/brake_release std_srvs/srv/Trigger "{}"
in terminal 2 " ros2 service call /io_and_status_controller/resend_robot_program std_srvs/srv/Trigger "{}"

## Expected Behavior

The robot_mode topic should show different values for different states (normal, running, pause).

## Actual Behavior
The robot_mode topic always shows mode: 7, regardless of the state (normal, running, pause).

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2024-11-27:
Hi, thank you for reporting this. The `robot_mode` will always be `RUNNING` as long as the brakes are released. That kind of corresponds to the teach pendant showing the green button in the corner (without the text next to it). Whether or not a program is running is independent of that. When booting the robot it will be landing in `POWER_OFF` and then you usually start the robot using the teach pendand cycling through `POWER_ON`, `IDLE` to `RUNNING`. See the [`RobotMode` message definition](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/e9499954614229f089a672b64cd63c1e2a49c328/ur_dashboard_msgs/msg/RobotMode.msg) for details.

If you want to check whether there is a program running, you would need the [`ProgramState`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/e9499954614229f089a672b64cd63c1e2a49c328/ur_dashboard_msgs/msg/ProgramState.msg). This would show whether there is currently *a* program running on the robot. As a matter of fact we currently do not expose that to ROS 2 through the driver.

But if I understand you correctly, you would like to know whether you can currently send a ROS trajectory to the robot, meaning that the `external control` program is running on the robot. For exactly this purpose there is the `/io_and_status_controller/robot_program_running`. It will tell you whether the external program is running and ready to receive commands. (In fact it will tell you whether the driver's `ReverseInterface` is connected, but unless you did something manual to change that behavior, the two statements should be equivalent in the context you asked for).

I hope, that helps.

bodkal, 2024-11-27:
Thanks, that's exactly what I needed!

urfeex, 2024-11-28:
Glad, I could help :-)

closed 2024-11-28 by urfeex