UniversalRobots/Universal_Robots_ROS2_Driver #1191: Dashboard Server remote mode

zkeyg, 2024-11-26:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

latest

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*
Is there a way to query if the dashboard server is in Remote mode?

## Issue details
*Detailed description help us understand the problem. Code are welcome!*

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
*What did you expect and why?*

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2024-11-27:
We currently don't have a direct service call to support this, but the dashboard server itself supports querying for remote control mode. For a full dashboard server documentation, please see https://www.universal-robots.com/articles/ur/dashboard-server-e-series-port-29999/

In short, you can make any dashboard call using the `raw_request` service:

```console
$ ros2 service call /dashboard_client/raw_request ur_dashboard_msgs/srv/RawRequest "query: 'is in remote control'"               [ur_rolling]
waiting for service to become available...
requester: making request: ur_dashboard_msgs.srv.RawRequest_Request(query='is in remote control')

response:
ur_dashboard_msgs.srv.RawRequest_Response(answer='true')
```

zkeyg, 2024-11-27:
Thank you @urfeex This is truly very helpful.

urfeex, 2024-11-28:
Glad I could help

closed 2024-11-28 by urfeex