UniversalRobots/Universal_Robots_ROS2_Driver #1192: external control error

Aruen24, 2024-11-27:
### Feature summary

velocity 54 required to reach the received target within 0.002 seconds is exceeding the joint velocity limits

urfeex, 2024-11-27:
Thank you for your report. Some context could be very helpful with this issue.

- What motion did you send to the robot with which controller?
- Did the program recover from this?
- What kind of ROS system do you use? -
  - Do you use a PREEMPT_RT patched kernel or lowlatency or standard kernel?
  - Do you run natively on a linux machine at all?
- What is the network connection to the robot?
  - Direct cable?
  - Switch involved?

urfeex, 2024-11-28:
Could you try using the scaled_joint_trajectory_controller and post the exact trajectory message that is sent to the controller?

bodkal, 2025-01-05:
I have the same problem.
 It looks like it is trying to reach the last point in the trajectory in 0.002 seconds (1.8784, -3.1874, -0.8803, -0.6447, 1.5708, 0.3076)."
[image]

im using ur20 with the sim docker and ROS2 Humble Ubuntu 22.04 standard kernel.

controllers: 

> picko_koby@pickokoby:~$ ros2 control list_controllers
> scaled_joint_trajectory_controller ur_controllers/ScaledJointTrajectoryController                active  
> io_and_status_controller           ur_controllers/GPIOController                                 active  
> speed_scaling_state_broadcaster    ur_controllers/SpeedScalingStateBroadcaster                   active  
> force_torque_sensor_broadcaster    force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster  active  
> ur_configuration_controller        ur_controllers/URConfigurationController                      active  
> tcp_pose_broadcaster               pose_broadcaster/PoseBroadcaster                              active  
> joint_state_broadcaster            joint_state_broadcaster/JointStateBroadcaster                 active  
> forward_position_controller        position_controllers/JointGroupPositionController             inactive
> force_mode_controller              ur_controllers/ForceModeController                            inactive
> freedrive_mode_controller          ur_controllers/FreedriveModeController                        inactive
> passthrough_trajectory_controller  ur_controllers/PassthroughTrajectoryController                inactive

this is the data in the trajectory msg:
```
trajectory.joint_trajectory.header.stamp.sec: 0
trajectory.joint_trajectory.header.stamp.nanosec: 0
trajectory.joint_trajectory.header.frame_id: world

current joints: 0.3169 -1.9647 -2.0930 -0.6548 1.5708 -2.8247

point: 0
positions: , 0.3169, -1.9647, -2.0930, -0.6548, 1.5708, -2.8247
velocities: , -0.0000, 0.0000, -0.0000, -0.0000, 0.0000, 0.0000
accelerations: , 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000
point: 1
positions: , 0.3147, -1.9633, -2.0933, -0.6558, 1.5708, -2.8197
velocities: , -0.0429, 0.0280, -0.0068, -0.0212, 0.0000, 0.1000
accelerations: , -0.4294, 0.2800, -0.0678, -0.2122, 0.0000, 1.0000
point: 2
positions: , 0.3083, -1.9591, -2.0943, -0.6590, 1.5708, -2.8047
velocities: , -0.0863, 0.0558, -0.0129, -0.0429, 0.0000, 0.2000
accelerations: , -0.4387, 0.2737, -0.0503, -0.2235, 0.0000, 1.0000
point: 3
positions: , 0.2974, -1.9521, -2.0958, -0.6645, 1.5708, -2.7797
velocities: , -0.1308, 0.0827, -0.0166, -0.0660, 0.0000, 0.3000
accelerations: , -0.4527, 0.2635, -0.0233, -0.2402, 0.0000, 1.0000
point: 4
positions: , 0.2821, -1.9426, -2.0975, -0.6723, 1.5708, -2.7447
velocities: , -0.1769, 0.1082, -0.0171, -0.0911, 0.0000, 0.4000
accelerations: , -0.4713, 0.2470, 0.0155, -0.2625, 0.0000, 1.0000
point: 5
positions: , 0.2620, -1.9306, -2.0991, -0.6828, 1.5708, -2.6997
velocities: , -0.2252, 0.1318, -0.0131, -0.1187, 0.0000, 0.5000
accelerations: , -0.4942, 0.2234, 0.0663, -0.2897, 0.0000, 0.9999
point: 6
positions: , 0.2369, -1.9163, -2.0999, -0.6961, 1.5708, -2.6447
velocities: , -0.2759, 0.1525, -0.0034, -0.1492, 0.0000, 0.6000
accelerations: , -0.5191, 0.1886, 0.1306, -0.3192, 0.0000, 1.0000
point: 7
positions: , 0.2067, -1.9002, -2.0995, -0.7127, 1.5708, -2.5797
velocities: , -0.3291, 0.1692, 0.0135, -0.1827, 0.0000, 0.7000
accelerations: , -0.5445, 0.1420, 0.2080, -0.3500, 0.0000, 1.0000
point: 8
positions: , 0.1711, -1.8827, -2.0970, -0.7328, 1.5708, -2.5047
velocities: , -0.3846, 0.1804, 0.0387, -0.2191, 0.0000, 0.8000
accelerations: , -0.5647, 0.0782, 0.2981, -0.3763, 0.0000, 0.9999
point: 9
positions: , 0.1297, -1.8643, -2.0915, -0.7566, 1.5708, -2.4197
velocities: , -0.4419, 0.1844, 0.0735, -0.2579, 0.0000, 0.8999
accelerations: , -0.5777, -0.0020, 0.3999, -0.3979, 0.0000, 0.9999
point: 10
[... 295 lines cut ...]
point: 84
positions: , 1.8784, -3.1028, -0.9653, -0.6442, 1.5708, 0.3076
velocities: , 0.0000, -0.4101, 0.3866, 0.0235, 0.0000, 0.0000
accelerations: , -0.0000, 1.0006, -0.6941, -0.3065, 0.0000, -0.0000
point: 85
positions: , 1.8784, -3.1388, -0.9304, -0.6431, 1.5708, 0.3076
velocities: , 0.0000, -0.3102, 0.3089, 0.0013, 0.0000, 0.0000
accelerations: , -0.0000, 1.0006, -0.8536, -0.1470, 0.0000, -0.0000
point: 86
positions: , 1.8784, -3.1648, -0.9040, -0.6435, 1.5708, 0.3076
velocities: , 0.0000, -0.2102, 0.2174, -0.0072, 0.0000, 0.0000
accelerations: , -0.0000, 1.0004, -0.9688, -0.0317, 0.0000, -0.0000
point: 87
positions: , 1.8784, -3.1809, -0.8872, -0.6443, 1.5708, 0.3076
velocities: , 0.0000, -0.1115, 0.1180, -0.0065, 0.0000, 0.0000
accelerations: , -0.0000, 0.9625, -0.9998, 0.0373, 0.0000, -0.0000
point: 88
positions: , 1.8784, -3.1873, -0.8804, -0.6446, 1.5708, 0.3076
velocities: , 0.0000, -0.0170, 0.0180, -0.0010, 0.0000, 0.0000
accelerations: , -0.0000, 0.9443, -1.0000, 0.0557, 0.0000, -0.0000
point: 89
positions: , 1.8784, -3.1874, -0.8803, -0.6447, 1.5708, 0.3076
velocities: , -0.0000, 0.0000, -0.0000, 0.0000, -0.0000, -0.0000
accelerations: , -0.0000, 0.9443, -1.0000, 0.0557, -0.0000, -0.0000
```

urfeex, 2025-01-08:
@bodkal thank you for posting an example path. Would you also have a time parametrization for all the trajectory points? That would make it a reproducible trajectory.

kdeshp-prof, 2025-01-09:
I am able to observe the same error even with URSim while using the ur ros2 driver (humble/Jazzy)

mohd-osama-47, 2025-01-16:
I have commented on a similar issue (https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1235#issuecomment-2594605381) a possible solution through lowering the speed slider in the UR Teach Pendant, not sure why this worked for me though...

bodkal, 2025-01-16:
I found out that if I add more points to the path (every 0.005 in the Cartesian path) and wait a bit before executing the path after the arm moves (maybe I send a new path before the first one fully converges?. Is there an API that can tell me if asynchronous paths are finished?), I stop getting that error.

urfeex, 2025-01-29:
@bodkal could you please open a new issue with your specific problem? We would need a specific way to reproduce it in order to debug things.

In the meantime, you could try building the client library and driver from source, using [this branch](https://github.com/urfeex/Universal_Robots_Client_Library/tree/improve_limit_check) (from my fork) for the client library? 

@Aruen24 we would need further information on how to reproduce this. Otherwise there is not much we can do to fix this.

urfeex, 2025-03-24:
We have updated the client library with regards to the limit check recently. Please try again with an up-to-date ur_client_library (>= 1.7.0).

urfeex, 2025-06-25:
@s4lmon As this thread is rather old and not providing enough information, would you please create a new issue describing your situation and a way to reproduce the problems you are facing?