UniversalRobots/Universal_Robots_ROS2_Driver #1201: Issues with controller disconnection when using set_digital_out and /urscript_interface on a UR3

SebastianBurbano, 2024-12-04:
### Affected ROS2 Driver version(s)

Humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

lastest

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

Hi, I’m working with a UR3 and the official ur_robot_driver. My project involves executing trajectories with the scaled_joint_trajectory_controller, and after completing some trajectories, I need to open or close a gripper using the set_digital_out command via the /urscript_interface/script_command topic.

My trajectory node is based on the example_move.py provided in this repository, which uses jtc_client to execute trajectories.

However, I have observed the following issues:

    1.Controller disconnection:  When publishing to /urscript_interface/script_command, the driver controller disconnects, and I see the following message:

    [UR_Client_Library]: Connection to reverse interface dropped.

     2. Issues with reconnection: My solution attempts to reconnect the controller by calling the /dashboard_client/play service. While the logger indicates a successful reconnection, subsequent trajectory executions within the same node no longer return a SUCCESSFUL result. This causes the node to hang and not complete the remaining trajectories.

Current Workflow:

1. I execute a trajectory with the scaled_joint_trajectory_controller.
2.  After completing the trajectory, I send a command to the gripper:
    ros2 topic pub --once /urscript_interface/script_command std_msgs/msg/String '{data: "def my_prog(): set_digital_out(0, True) end"}'

3. After the driver disconnects, I call a method to reconnect the driver using the /dashboard_client/play service. However, after this, I can no longer execute further trajectories within the same node.

Question:
Is there a recommended way to use set_digital_out or control the gripper without causing the controller to disconnect? Is there any additional configuration needed in the driver or the control node to handle this scenario?

I would greatly appreciate any guidance or solutions on managing the gripper without interrupting the trajectories. Thank you!

If you have any doubts about how my node works, you can check it on my GitHub: https://github.com/SebastianBurbano/ur3_packaging_control/blob/main/src/control_ur3/control_ur3/empaque_prueba.py

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2024-12-05:
Hi, thank you for reaching out! May I suggest using the [gpio controller](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_robot_driver/ur_controllers/doc/index.html#advertised-services) for setting digital outputs?

```
ros2 service call /io_and_status_controller/set_io ur_msgs/srv/SetIO "{fun:` 1, pin: 0, state: 1.0}"
```

However, the hanging issue that you mentioned should not happen. Restarting the program which reconnects the reverse interface should result in an operational driver.

SebastianBurbano, 2024-12-05:
Hi, thank you for your response. I have adjusted my node to control the gripper using the GPIO controller as you suggested. However, the problem persists, although now the connection to the driver doesn't drop. You can see more details about what happens in the following images:
[image]
[image]

In the YAML file containing the trajectories, it can be seen that after traj2, there is another trajectory that is not executed, as well as another gripper action that is also not executed.

[image]

urfeex, 2025-02-06:
Sorry for coming back to this so late. As far as I understand your screenshots, traj2 finishes in the controller, but your calling node never gets notified of that and thus doesn't succeed?

That could also be due to an error in your client code, though. As the changed code seems not to have landed in the one you linked earlier, I can not verify that at the moment.

urfeex, 2025-03-24:
Closing this as no further feedback has been provided. Feel free to comment / reopen if new input exists.

closed 2025-03-24 by urfeex