UniversalRobots/Universal_Robots_ROS2_Driver #1205: Connection could not be established.

Ryan-Lockley, 2024-12-16:
### Affected ROS2 Driver version(s)

2.4.13

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Windows using WSL2

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

PolyScope 74.11.1642

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Hello. I am new to robotics and Linux, so this may be easily solved.
I am trying to connect a UR10e arm to my laptop to control it.
However, I can't seem to get past "Receive Program Failed: The connection to the remote PC at <address:port> could not be established. Reason: connect timed out"
I can see the robot position in RViz, it updates if I control it via the Teach Pendant. I can ping the robot and get a response.
But I cannot connect the Teach Pendant to my computer.

## Issue details
Despite the addresses of the laptop and the arm seeming to be correct, there is no connection on the Teach Pendant. I can however see the arm position updating in RViz and ping the robot.

## Steps to Reproduce
Run in Ubuntu window 1: 

_ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.56.101 launch_rviz:=false headless_mode:=true_

Run in Ubuntu window 2:

_ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10e launch_rviz:=true_

## Expected Behavior
I think the next step is for me to run the program "Control by <laptop name>" and then I can send commands in RViz/MoveIt to move the arm.

## Actual Behavior
When tapping on Run:
"Receive Program Failed: The connection to the remote PC at <address:port> could not be established. Reason: connect timed out"
In settings: "Not connected to network!"
Ubuntu window 1:
_[ros2_control_node-1] [INFO] [1734354802.007527679] [scaled_joint_trajectory_controller]: Received new action goal
[ros2_control_node-1] [INFO] [1734354802.007594279] [scaled_joint_trajectory_controller]: Accepted new action goal_
Ubuntu window 2:
_[move_group-2] [INFO] [1734354885.789008521] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[move_group-2] [INFO] [1734354885.789874121] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-2] [INFO] [1734354885.789907521] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!_

# Workaround Suggestion
None yet.

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2024-12-18:
The popup on the teach pendant should give you  a specific IP address and port (what you reported as `address:port`).

- Is that IP address the one of your PC running the ROS driver?
- Can the robot reach that IP address?
- Is there a firewall blocking access to the port number?
- Since you are running WSL, that raises a couple of additional questions:
  - Does the linux container know the IP address the robot os connecting to? If that is the IP address of your windows host, the WSL container might not know it / bind sockets to it.
  - You probably have to specify your [reverse_ip](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/8d4ba9421cce73fbade5286be574640e6f6c7326/ur_robot_driver/launch/ur_control.launch.py#L431)
  - maybe [this post](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/486#issuecomment-1096776059) does help?

Ryan-Lockley, 2024-12-18:
Thank you for your help.
I did have a look at some of your points.
But in the interest of time, we opted to use a native Ubuntu machine which, ultimately just worked.

I believe the issue may have been with our IT security policies. That or the IP that Linux was using not matching what Windows was using, I'm not sure as I'm a novice at this.

Either way it's working on a different machine now.

closed 2024-12-19 by urfeex