UniversalRobots/Universal_Robots_ROS2_Driver #1206: Test Fail Locally

erichlf, 2024-12-16:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

Proxmox 5.11.11

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
I am attempting to run the ur ros2 driver tests. But the tests all fail.

## Issue details
 I have tried a combination of things, but they all involve docker. I have tried running URSim in its own docker container with the driver tests running in another  docker container, and I have tried running a docker container for the driver and then issuing `ros2 run ur_robot_client start_usim.sh`, but I get the same result.

## Steps to Reproduce
First thing is to clone the appropriate repos into ${HOME}/workspace/ur/src. Then I have the following `.env`
```
PUID=${PUID}
PGID=${PGID}
HOME=${HOME}
WORKSPACE="${HOME}/workspace/ur"
POLYSCOPE=5.11.11
URSIM_WS="${WORKSPACE}"
```
and the following docker-compose file
```yaml
services:
  ursim:
    image: universalrobots/ursim_e-series:${POLYSCOPE}
    container_name: ursim
    ports:
      - 5900:5900
      - 6080:6080
      # added
      - 29999:29999
      - 30001:30001
      - 30002:30002
      - 30004:30004
      - 50002:50002
    environment:
      ROBOT_MODEL: UR10
      TZ: Europe/Amsterdam
    volumes:
      - ${URSIM_WS}/.ursim/urcaps:/ursim/.urcaps
      - ${URSIM_WS}/.ursim/programs:/ursim/programs
      - ${URSIM_WS}/.ursim/gui:/ursim/GUI
    networks:
      ursim_net:
        ipv4_address: 192.168.56.101

  coresense:
    build: .
    env_file: variables.env
    user: ${PUID:-1000}:${PGID:-1000}
    container_name: coresense-test
    tty: true
    cap_add:
      - SYS_PTRACE
    security_opt:
      - seccomp:unconfined
      - apparmor:unconfined
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix
      - ${HOME}/.Xauthority:/home/ros/.Xauthority:rw
      - /dev:/dev
      # issue where opencv can't display an image
      - /run/user/${PUID:-1000}/at-spi/bus_0:/run/user/${PUID:-1000}/at-spi/bus_0:ro
      - ${WORKSPACE}:/home/ros/ros_ws
    group_add:
      - video
      - tty
    privileged: true
    devices:
      - /dev/dri
    networks:
      - ursim_net
    command: bash

networks:
  ursim_net:
    driver: bridge
    ipam:
      config:
        - subnet: 192.168.56.0/24
```
with `Dockerfile`:
```
ARG ROS_DISTRO=humble
ARG PREFIX=

FROM registry.tudelft.nl/samxl/moveit2:master-on-humble-source2 AS ros_user_setup
# Create a non-root user
ENV ROS_USER=ros
ARG ROS_USER_UID=1000
ARG ROS_USER_GID=$ROS_USER_UID

RUN groupadd --gid $ROS_USER_GID $ROS_USER && \
  useradd -s /bin/bash --uid $ROS_USER_UID --gid $ROS_USER_GID -m $ROS_USER \
  # [Optional] Add sudo support for the non-root user
  && apt-get update \
  && apt-get install -y --no-install-recommends sudo \
  && echo $ROS_USER ALL=\(root\) NOPASSWD:ALL > /etc/sudoers.d/$ROS_USER \
  && chmod 0440 /etc/sudoers.d/$ROS_USER

RUN apt-get update \
  # install build tools
  && sudo apt-get install -y --no-install-recommends \
    alsa-utils \
    apt-rdepends \
    build-essential \
    git \
    # install useful tools
    ccache \
    wget \
    curl \
    htop \
    vim \
    clang-format \
    # install ros specifics
    libpoco-dev \
    libprotobuf-dev \
    protobuf-compiler \
    python3-colcon-common-extensions \
    python3-pip \
    python3-rosdep \
    python3-vcstool \
  # apt clean-up
  && sudo apt-get autoremove -y \
  && sudo rm -rf /var/lib/apt/lists/*

USER $ROS_USER

# Set up the entrypoint
ENTRYPOINT [ "/ros_entrypoint.sh" ]
CMD ["bash"]
```
Then I run 
1. `docker compose up` everything comes up and I get a link that point to the URSim.
2. `docker exec -it coresense-test bash`
3.  `source /opt/ros/humble/setup.bash`
4. `cd ~/ros_ws`
5. `rosdep update && rosdep -q install --from-paths src/ --ignore-src -y --rosdistro "${ROS_DISTRO}"`
6. `colcon build --symlink-install --merge-install --packages-up-to ur --cmake-args -DUR_ROBOT_DRIVER_BUILD_INTEGRATION_TESTS=on`
7. `source install/setup.bash`
8. `colcon test --merge-install --event-handlers console_cohesion+ --packages-select ur_robot_driver`

## Expected Behavior
I would have expected that the tests pass.

## Actual Behavior
All tests fail

# Workaround Suggestion
No idea

### Relevant log output

```shell
The following tests FAILED:
	  1 - test_launch_args.py (Failed)
	  2 - test_dashboard_client.py (Failed)
	  3 - test_robot_driver.py (Failed)
	  4 - test_integration_test_controller_switch.py (Failed)
	  5 - test_urscript_interface.py (Failed)
```

### Accept Public visibility

- [X] I agree to make this context public

erichlf, 2024-12-16:
For the version where I just run the test and don't bring up URsim separately I get the following logs:
[test_no_ursim.log](https://github.com/user-attachments/files/18151484/test_no_ursim.log)

urfeex, 2024-12-16:
The problematic line from your output could be 

```
1: [ur_ros2_control_node-3] message: SafetySetup has not been confirmed yet, therefore no data will be send yet
```

Running your example leaves me with a non-running ursim container:

```
[+] Running 3/3
 ✔ Network docker_test_ursim_net  Created                                                                                                               0.1s 
 ✔ Container ursim                Created                                                                                                               0.0s 
 ✔ Container ur-driver-test       Created                                                                                                               0.0s 
Attaching to ur-driver-test, ursim
ursim           | Universal Robots simulator for e-Series:5.11.11
ursim           | 
ursim           | 
ursim           | Access the robots user interface through this URL:
ursim           | 
ursim           |      http://localhost:6080/vnc.html?host=localhost&port=6080
ursim           | 
ursim           | 
ursim           | Access the robots user interface with a VNC application on this address:
ursim           | 
ursim           |      localhost:5900
ursim           | 
ursim           | 
ursim           | You can find documentation on how to use this container on dockerhub:
ursim           | 
ursim           |      https://hub.docker.com/r/universalrobots/ursim_e-series
ursim           | 
ursim           | 
ursim           | Press Crtl-C to exit
ursim           | 
ursim           | 
ursim           | 
ursim           | 
ursim exited with code 0
```

erichlf, 2024-12-16:
That is strange, because I have run this multiple times and it runs URSim and I can connect to the posted URL.

urfeex, 2024-12-16:
Removing the `${URSIM_WS}/.ursim/gui:/ursim/GUI ` volume from the compose file makes things start for me and the tests also run successfully for me.

Since a similar setup is also running in CI where the `start_ursim` script launches ursim in another docker container, I'll leave this as non-reproducible for now.

erichlf, 2024-12-16:
> Removing the `${URSIM_WS}/.ursim/gui:/ursim/GUI ` volume from the compose file makes things start for me and the tests also run successfully for me.
>
> Since a similar setup is also running in CI [...]

So you start URSim and then run the tests?

urfeex, 2024-12-16:
> So you start URSim and then run the tests?

Yes, using the commands you listed above (skipping the docker installation but adding a `sudo apt update` at that point)

closed 2024-12-16 by erichlf