UniversalRobots/Universal_Robots_ROS2_Driver #1223: Add the robot_calibration.yaml in the URcap

LucaMIMOUNI, 2025-01-06:
### Feature summary

Hey everyone,

I've did the calibration for UR10e following this tutorial: [calib](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/robot_setup.html#extract-calibration-information). 

I want to know **if there is a way to include this calibration file directly 'in the robot'**. For exemple, I'm using it when launching the ROS2 driver, but if now I want to use RoboDK, I need to calibrate it on RoboDK, ect ...

_**The simplest way would be to directly plug an USB in the URcap and put the calibration directly in the robot**_, so that I can use any software I want, it would already be calibrate.

If someone could help me on this, would be much appreciated !

urfeex, 2025-01-08:
The calibration extraction is specific to ROS / the urdf representation in [ur_description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description).

The calibration itself is stored on the robot already. The calibration extraction process is reading the calibration information from the robot and creates a set of parameters to parametrize the robot description as mentioned above.

If you want to use another software with the robot you'll have to make sure that the representation used is utilizing the robot's calibration. E.g. the calibration process prints the robot's DH parameters that you can build your model upon. This might not directly be usable depending on what you want to do with the robot. See the [algorithm description](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_calibration/doc/algorithm.html) for what we do with these parameters to get our representation.

closed 2025-01-08 by urfeex

LucaMIMOUNI, 2025-01-08:
You talked about DH parameters, but the calibration from [calib](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/robot_setup.html#extract-calibration-information) give a yaml file with XYZ, Roll-Pitch-Yaw for each joints instead. 

How can I get the DH parameters from the URcap as the ros command give me something else ?

urfeex, 2025-01-09:
You will get the parameters from the output of the calibration extraction script:

```
$ ros2 launch ur_calibration calibration_correction.launch.py robot_ip:=192.168.56.101                                                                                                                                                                                                          [ur_rolling]
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [calibration_correction-1]: process started with pid [942610]
[calibration_correction-1] [WARN 2025-01-09 08:56:20.469] [UR_Client_Library:]: Failed to read from stream, reconnecting in 1 seconds...
[calibration_correction-1] [INFO 2025-01-09 08:56:21.507] [ur_calibration]: checksum: [4294967295 4294967295 4294967295 4294967295 4294967295 4294967295 ]
[calibration_correction-1] dh_theta: [0 0 0 0 0 0 ]
[calibration_correction-1] dh_a: [0 -0.425 -0.3922 0 0 0 ]
[calibration_correction-1] dh_d: [0.1625 0 0 0.1333 0.0997 0.0996 ]
[calibration_correction-1] dh_alpha: [1.570796327 0 0 1.570796327 -1.570796327 0 ]
[calibration_correction-1] calibration_status: 0
```

Note: These parameters are from a simulated robot, real values will have more non-zero entries.