UniversalRobots/Universal_Robots_ROS2_Driver #1228: Driver on jazzy doesn't work when using use_mock_hardware flag

PietroValerio, 2025-01-11:
### Affected ROS2 Driver version(s)

3.0.0-1noble.20241228.020252 amd64

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot, Real robot, URSim in docker

### Robot SW / URSim version(s)

5.19

### How is the ROS driver used.

Others

### Issue details

# Summary
When launching the launchfile for starting the controller, i get all the messages reported below, also trying to start moveit servo via their cpp api, doesn't work because you get this message [scaled_joint_trajectory_controller]: Holding position due to state tolerance violation when the code used to work, probably on the older version of the binary of which i don't remember the version, but i can verify this only from monday

## Steps to Reproduce
Run the driver launch file with the flag use_mock_hardware:=true, also try it on the example cell configuration https://github.com/UniversalRobots/Universal_Robots_ROS2_Tutorials you get the same errors

## Expected Behavior
No errors because in the last version my code worked in simulation and i tested it on real hardware (ur10e)

## Actual Behavior
The errors shown below and servoing not working

### Relevant log output

```shell
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10e robot_ip:=192.166.43.22 use_mock_hardware:=true launch_rviz:=false
[INFO] [launch]: All log files can be found below /home/pietro/.ros/log/2025-01-11-22-09-23-432808-pietro-86197
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [86201]
[INFO] [robot_state_publisher-2]: process started with pid [86202]
[INFO] [spawner-3]: process started with pid [86203]
[INFO] [spawner-4]: process started with pid [86204]
[robot_state_publisher-2] [INFO] [1736629763.730426005] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1736629763.749630194] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-1] [INFO] [1736629763.749740622] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1736629763.749746503] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1736629763.749822416] [controller_manager]: Could not enable FIFO RT scheduling policy: with error number <1>(Operation not permitted). See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[ros2_control_node-1] [INFO] [1736629763.759884468] [controller_manager]: Received robot description from topic.
[ros2_control_node-1] [INFO] [1736629763.764349807] [controller_manager.resource_manager]: Loading hardware 'ur10e' 
[ros2_control_node-1] [INFO] [1736629763.765096996] [controller_manager.resource_manager]: Loaded hardware 'ur10e' from plugin 'mock_components/GenericSystem'
[ros2_control_node-1] [INFO] [1736629763.765131582] [controller_manager.resource_manager]: Initialize hardware 'ur10e' 
[ros2_control_node-1] [INFO] [1736629763.765350925] [controller_manager.resource_manager]: Successful initialization of hardware 'ur10e'
[ros2_control_node-1] [WARN] [1736629763.765462135] [controller_manager.resource_manager]: Importing state interfaces for the hardware 'ur10e' returned no state interfaces.
[ros2_control_node-1] [INFO] [1736629763.765667692] [resource_manager]: 'configure' hardware 'ur10e' 
[ros2_control_node-1] [INFO] [1736629763.765674185] [resource_manager]: Successful 'configure' of hardware 'ur10e'
[ros2_control_node-1] [INFO] [1736629763.765701456] [resource_manager]: 'activate' hardware 'ur10e' 
[ros2_control_node-1] [INFO] [1736629763.765705544] [resource_manager]: Successful 'activate' of hardware 'ur10e'
[ros2_control_node-1] [INFO] [1736629763.765716484] [controller_manager]: Resource Manager has been successfully initialized. Starting Controller Manager services...
[ros2_control_node-1] [INFO] [1736629763.969198314] [controller_manager]: Loading controller : 'joint_trajectory_controller' of type 'joint_trajectory_controller/JointTrajectoryController'
[ros2_control_node-1] [INFO] [1736629763.969232519] [controller_manager]: Loading controller 'joint_trajectory_controller'
[... 162 lines cut ...]
[ros2_control_node-1] [ERROR] [1736629764.242322626] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.243969381] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.243980492] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.246166614] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.246188094] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.248403271] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.248420023] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.250065645] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.250078259] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.252296221] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.252308424] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.253962082] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.253971901] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.256165106] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.256177109] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.258391634] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[... cut ...]
[ros2_control_node-1] [ERROR] [1736629764.260058988] []: TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of an invalid quaternion in the transform (nan nan nan nan)
[ros2_control_node-1] [ERROR] [1736629764.262278072] []: TF_NAN_INPUT: Ignoring transform for child_frame_id "tool0_controller" from authority "Authority undetectable" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
```

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2025-01-13:
Thank you for posting this. We are in fact currently working on this. #1226 disables launching the pose_broadcaster (which was producing this error output) and should help finding other things that don't work with mock_hardware.

In the meantime we've added some documentation listing non-working features in mock_hardware in #1227.

@URJala will create a separate PR having https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/1226/files#diff-5fcec6b0bfc65068dc83b8bbb2bce993d9cac3a54be670dcee8be6fb1b70e2afR185-R187 available earlier which should prevent the issue you've been experiencing.

PietroValerio, 2025-01-13:
Thanks! If it can be helpful i've noticed that on 2.4.13 was possible to make servoing (simulated and real configuration) via MoveIt2 cpp api, sending the commands directly to the `scaled_joint_trajectory_controller`, moveit gives the possibility to generate a `trajectory_msgs::msg::JointTrajectory`  using `composeTrajectoryMessage()`. In the current version of the driver is not possible to do this because you get from the driver `scaled_joint_trajectory_controller]: Holding position due to state tolerance violation`, instead using `forward_velocity_controller` makes the servo work, but in my opinion it was simpler and more easily adaptable to use in any case `scaled_joint_trajectory_controller` to compose planned movements with servo movements

closed 2025-01-13 by urfeex