UniversalRobots/Universal_Robots_ROS2_Driver #1235: Velocity 3141.6005 required to reach the received target within 0.002 seconds is exceeding the joint velocity limits.

rodrigoantunes2000, 2025-01-14:
### Affected ROS2 Driver version(s)

Ubuntu 24.04 ros-jazzy-ur

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

ur3e

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
When I use the ur_moveit_config.py and move the robot through the rviz (for example rotate the first joint 5 degrees), the "Velocity 3141.6005 required to reach the received target within 0.002 seconds is exceeding the joint velocity limits" error appears

## Issue details

## Steps to Reproduce
I run the command: ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.1.102 launch_rviz:=false kinematics_params_file:="${HOME}/my_robot_calibration.yaml"

And then, in other command window I run: ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true

The communication appears to be working fine because if I move the robot in freedrive, the rviz is updating the pose.

## Expected Behavior
When I move the robot with move_it, just a simple 5 degree movement with the first joint, I expected the action to be planned easily using the ompl planning and executed

## Actual Behavior
The robot doesnt move and the error "Velocity 3141.6005 required to reach the received target within 0.002 seconds is exceeding the joint velocity limits" appears on the pendant. Additionaly, it only seems capable of planning when the option "Use cartesian path" is selected and in previous attempts (in early october 2024) I did plan without this option selected and it worked well in the execution of the task

# Workaround Suggestion
Every time I run: ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true

The are 4 warnings that tell something is not working properly regarding the robot urdf, something that I think it didnt happen in my first try back in october 2024:
WARNING:root:Cannot infer URDF from `/opt/ros/jazzy/share/ur_moveit_config`. -- using config/ur.urdf
WARNING:root:Cannot infer SRDF from `/opt/ros/jazzy/share/ur_moveit_config`. -- using config/ur.srdf
WARNING:root:"File /opt/ros/jazzy/share/ur_moveit_config/config/ur.urdf doesn't exist"
WARNING:root:The robot description will be loaded from /robot_description topic 

The full output of the command is:
[INFO] [launch]: All log files can be found below /home/ros2/.ros/log/2025-01-14-15-21-35-122479-ros2-5824
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/opt/ros/jazzy/share/ur_moveit_config`. -- using config/ur.urdf
WARNING:root:Cannot infer SRDF from `/opt/ros/jazzy/share/ur_moveit_config`. -- using config/ur.srdf
WARNING:root:"File /opt/ros/jazzy/share/ur_moveit_config/config/ur.urdf doesn't exist"
WARNING:root:The robot description will be loaded from /robot_description topic 
[INFO] [wait_for_robot_description-1]: process started with pid [5831]
[wait_for_robot_description-1] [INFO] [1736868096.310253235] [wait_for_robot_description]: Waiting for message on /robot_description.
[wait_for_robot_description-1] [INFO] [1736868097.168360282] [wait_for_robot_description]: Received message on /robot_description. Shutting down.
[INFO] [wait_for_robot_description-1]: process has finished cleanly [pid 5831]
[INFO] [move_group-2]: process started with pid [5848]
[INFO] [rviz2-3]: process started with pid [5849]
[... 134 lines cut ...]
[move_group-2] 
[move_group-2] You can start planning now!
[move_group-2] 
[rviz2-3] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-3]          at line 321 in /opt/ros/jazzy/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-3] [WARN] [1736868099.280048733] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2_moveit'. If this is due to multiple nodes with the same name then all logs for the logger named 'rviz2_moveit' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-3] [ERROR] [1736868102.391152214] [rviz2_moveit.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-3] [INFO] [1736868102.448842562] [rviz2_moveit.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-3] [WARN] [1736868102.628998756] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2_moveit'. If this is due to multiple nodes with the same name then all logs for the logger named 'rviz2_moveit' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-3] [INFO] [1736868102.833112734] [rviz2_moveit.moveit.ros.rdf_loader]: Loaded robot model in 0.208441 seconds
[rviz2-3] [INFO] [1736868102.833550039] [rviz2_moveit.moveit.core.robot_model]: Loading robot model 'ur3e'...
[rviz2-3] [INFO] [1736868102.833616985] [rviz2_moveit.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-3] [INFO] [1736868102.911920083] [rviz2_moveit.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'ur_manipulator': 1 1 1 1 1 1
[rviz2-3] [INFO] [1736868103.168012931] [rviz2_moveit.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[rviz2-3] [INFO] [1736868103.173763367] [rviz2_moveit.moveit.ros.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-3] [INFO] [1736868103.775763562] [interactive_marker_display_187651596814864]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-3] [INFO] [1736868103.807093407] [rviz2_moveit.moveit.ros.motion_planning_frame]: group ur_manipulator
[rviz2-3] [INFO] [1736868103.807164002] [rviz2_moveit.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'ur_manipulator' in namespace ''
[rviz2-3] [INFO] [1736868103.926604261] [interactive_marker_display_187651596814864]: Sending request for interactive markers
[rviz2-3] [INFO] [1736868103.991082565] [rviz2_moveit.moveit.ros.move_group_interface]: Ready to take commands for planning group ur_manipulator.
[rviz2-3] [INFO] [1736868104.083651774] [interactive_marker_display_187651596814864]: Service response received for initialization

### Accept Public visibility

- [X] I agree to make this context public

urfeex, 2025-01-15:
This seems to be a large overlap / duplicate with #1192. I am also not able to reproduce this using a rolling source build against a 5.19 URSim. I have been planning and executing different motions using MoveIt without any problems.

 I will try again later using a Jazzy binary build.

rodrigoantunes2000, 2025-01-15:
With the URSim I had no problems, but with the real robot (UR3e) I couldnt. I tried to run the test_scaled_joint_trajectory_controller and I moved the robot successfuly, I just cant do it with move_it

urfeex, 2025-01-15:
MoveIt or not shouldn't make any difference in terms of control and URSim or real robot shouldn't make any difference in terms of MoveIt.

I'm not saying that your issue is invalid, I'm just having a hard time identifying the problem.

rodrigoantunes2000, 2025-01-15:
I am not an expert at all, but it seems to me that MoveIt is not catching properly the robot URDF and SRDF and that may cause problems in planning the motion of the robot. Maybe its an installation problem from my side, although I tried in different devices (PC with Linux and two times in RaspberryPi5), if it helps I leave here the history of commands I used in the instalation of ros2 and the ur_driver:

sudo apt update && sudo apt upgrade
sudo apt install tmux terminator git wget curl nano python3 python3-pip python3-venv

locale
sudo apt install software-properties-common
sudo add-apt-repository universe
sudo apt update && sudo apt install curl -y
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null
sudo apt update && sudo apt install ros-dev-tools
sudo apt upgrade
sudo apt install ros-jazzy-desktop
sudo apt-get update
sudo apt-get install ca-certificates curl shellinabox
sudo install -m 0755 -d /etc/apt/keyrings
sudo curl -fsSL https://download.docker.com/linux/ubuntu/gpg -o /etc/apt/keyrings/docker.asc
sudo chmod a+r /etc/apt/keyrings/docker.asc
echo   "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/docker.asc] https://download.docker.com/linux/ubuntu $(. /etc/os-release && echo "$VERSION_CODENAME") stable" |   sudo tee /etc/apt/sources.list.d/docker.list > /dev/null
sudo apt-get update
sudo docker run hello-world
sudo groupadd docker
sudo usermod -aG docker $USER
usermod -aG docker $USER
sudo usermod -aG docker $USER
usermod -aG docker $USER
newgrp docker

sudo apt-get install ros-jazzy-ur

echo "
# ROS2
source /opt/ros/jazzy/setup.bash
" >> .bashrc

# Run on simulator
ros2 run ur_client_library start_ursim.sh -m ur3e
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 launch_rviz:=true
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.101 launch_rviz:=false
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true

#Run in the real robot
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.1.102 launch_rviz:=false
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true

urfeex, 2025-01-15:
> I am not an expert at all, but it seems to me that MoveIt is not catching properly the robot URDF and SRDF and that may cause problems in planning the motion of the robot

No, that's not the case. By default, MoveIt loads its own urdf and publishes that. We deliberately did not do that in our description package, but re-use the robot_description published by the robot_state_publisher. Hence the warnings at the beginning, as the default files that MoveIt would look for are not there. The "fallback" `WARNING:root:The robot description will be loaded from /robot_description topic` is actually the desired behavior here.

Regarding the SRDF we use a xacro version to only maintain one common srdf for all robots. For this we [pass that manually to the move_group](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/79252cf92b25aaaf398f6deab2d6b1ba8878aba5/ur_moveit_config/launch/ur_moveit.launch.py#L139) node for republishing.

The output you've posted doesn't seem unnormal.

rodrigoantunes2000, 2025-01-15:
Thank you for your explanation, now it makes more sense to me.
Anything I can do to help you find the solution, feel free to ask!

mohd-osama-47, 2025-01-16:
For me personally, I have encountered a similar issue and was able to find a temporary workaround by setting the speed slider on the teach pendant of the UR I was using (UR5e) to be around 30% and then running a moveit config for a custom arm with a robotiq gripper.

I am not aware why this fixed it but please try and check from your side!

urfeex, 2025-01-29:
@rodrigoantunes2000 Could you help providing a way to reproduce this? The velocity in your example boils down to do roughly 2*Pi in once cycle. As you are using a UR3e I wonder whether this has anything to do with the limit-less wrist3-joint.

Therefore, if you could provide us with the information:
 
* What command did you exactly execute? (Ideally the action goal sent to the controller, otherwise the exact target configuration setup in MoveIt)
* What is the starting configuration (joint positions) where your robot is when running your command.

I've furthermore prepared a branch that should give better feedback about the limit check: https://github.com/urfeex/Universal_Robots_Client_Library/tree/improve_limit_check

You would have to build the client library and driver from source:

* Remove the client library first (`sudo apt remove ros-jazzy-ur-client-library` -- this will remove the driver, as well)
* Build the client library and driver:
   ```
   export COLCON_WS=~/workspace/ros_ur_driver
   mkdir -p $COLCON_WS/src
   cd $COLCON_WS
   git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/ur_robot_driver
   git clone -b improve_limit_check https://github.com/urfeex/Universal_Robots_Client_Library.git src/ur_client_library
   rosdep update
   rosdep install --ignore-src --from-paths src -y
   colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
   source install/setup.bash
   ```

closed 2025-03-24 by urfeex