UniversalRobots/Universal_Robots_ROS2_Driver #1236: driver issue,

junjunkiki, 2025-01-15:
### Feature summary

I'm using Ubuntu 22.04, ros2 humble. Additionally, I don't have an actual ur3 robot, so I'm just trying to do a simulation in rviz2. (such as going to any location you want via python code)
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3 use_fake_hardware:=true This command lets you launch the ur3 robot in rviz2. If you plan, you can make a plan, but if you actually execute it, you can launch it 
""""
move_group-1] [INFO] [1736921478.332243080] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[rviz2-2] [INFO] [1736921478.333601585] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1736921478.333670611] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-1] [INFO] [1736921478.334017374] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1736921478.335848213] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1736921478.336214944] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1736921479.336361593] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1736921478.336247, but latest received state has time 0.000000.
[move_group-1] Check clock synchronization if your are running ROS across multiple machines!
[move_group-1] [WARN] [1736921479.336502013] [moveit_ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[move_group-1] [INFO] [1736921479.336608801] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-2] [INFO] [1736921479.337928361] [move_group_interface]: Execute request aborted
[rviz2-2] [ERROR] [1736921479.338112590] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached"""" 
You get an error like that.
Chtgpt seems to be a controller problem.
I've tried several solutions I got from chatgpt, but I can't solve them.

colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
at the time of
2 packages hadstderr output: ur_controllers ur_robot_driver warning. Could this have any impact? 
 
How do I float the ur3 robot in rviz2 and perform the action using python?

urfeex, 2025-01-15:
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3 use_fake_hardware:=true` alone doesn't simulate a robot. It just launches MoveIt and RViz.

You'll also have to launch a simulated robot using 

```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e use_fake_hardware:=true robot_ip:=irrelevant
```

Note however: Your issue revealed to me that on Humble we were missing dropping the remap of the default controller. This will be fixed in #1237. For the time being, simply drop the `use_fake_hardware:=true` from your MoveIt startup.

junjunkiki, 2025-01-16:
Thank you for your answer.
I want to take a floating ur3e robot arm and see how the robot arm moves using moveit2. What should I do?
The code ros2launchur_robot_driver_control.launch.py ur_type:=ur3euse_fake_hardware:=true robot_ip:=irrelevant. I don't know how to move the robot arm with this

urfeex, 2025-01-16:
You basically need to do both:

In one shell: 
```ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e use_fake_hardware:=true robot_ip:=irrelevant```

In another shell: 
```ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e```

junjunkiki, 2025-01-17:
thank you for answer

ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e use_fake_hardware:=true robot_ip:=irrelevant
This lunch is looking at ur3e.
And
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e
Is this launch moving ur3e through moveit function?

So ros2 launcher ur_moveit_config ur_moveit.launch.py ur_type:=ur3e Is it correct that there was an error when I ran it alone??

urfeex, 2025-01-17:
> So ros2 launcher ur_moveit_config ur_moveit.launch.py ur_type:=ur3e Is it correct that there was an error when I ran it alone??

That's what I was trying to say, yes. MoveIt can be used either with a simulated robot or a real robot, it doesn't matter. It will connect to the one running. So, in order to be able to use MoveIt, you'll have to start a mocked, simulated or real robot  separately.

It seems that this is resolved, so I will close this issue. Feel free to comment / reopen if you disagree.

closed 2025-01-17 by urfeex