UniversalRobots/Universal_Robots_ROS2_Driver #1246: 404 on link to ROS2 humble build instructions

BeFranke, 2025-01-20:
### Affected ROS2 Driver version(s)

humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

n/a

### How is the ROS driver used.

Others

### Issue details

# Summary
It seems the link https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/installation/installation.html is dead, which is very unfortunate for anyone trying to build the humble version of the driver.

Thank you for providing this project!

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-01-21:
Thank you for pointing that out!

I guess, the content you are looking for is on https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/doc/installation/installation.html?

I further assume, you are referring to the link on this repo's README? Or did you come from another place we should update?

BeFranke, 2025-01-23:
Thank you for posting the correct link! 
Yes, the link was on the repos README (on the humble branch). My bad, I should have specified this in the issue.

urfeex, 2025-01-23:
Then we should have this covered by #247. Thanks again for pointing that out!

closed 2025-01-30 by urfeex