UniversalRobots/Universal_Robots_ROS2_Driver #1250: scaled_joint_trajectory_controller does not behave as expected when stop is pressed on the robot

URJala, 2025-01-24:
### Affected ROS2 Driver version(s)

3.0.2

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

Real robot, UR E-series robot, URSim in docker

### Robot SW / URSim version(s)

5.20

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
If the stop button is pressed on the teach pendant (real or simulated) while the scaled joint trajectory controller is executing a trajectory, the controller will flood the terminal running the ROS driver with an error message, and the ROS action responsible for the trajectory will never finish.

## Issue details
The error message reported is: "[ros2_control_node-1] [WARN] [1737719032.541174555] [realtime_tools]: goal_handle attempted invalid transition from state EXECUTING with event CANCELED, at ./src/rcl_action/goal_handle.c:95". 

## Steps to Reproduce
launch the driver with either a real robot or a dockersim attached and execute a trajectory on the scaled_joint_trajectory_controller. While it is executing, press stop on the robot teach pendant. 

## Expected Behavior
The controller aborts the trajectory and reports the ROS action as aborted to the user. It also does not flood the terminal with error messages.

## Actual Behavior
ROS action is never reported as finished, and the terminal is flooded with error messages. 

### Relevant log output

```shell
[ros2_control_node-1] [WARN] [1737719032.541174555] [realtime_tools]: goal_handle attempted invalid transition from state EXECUTING with event CANCELED, at ./src/rcl_action/goal_handle.c:95
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-02-03:
This seems to be an issue in the upstream JTC. I've submitted a possible fix in https://github.com/ros-controls/ros2_controllers/pull/1517

urfeex, 2025-02-04:
I manually checked that the upstream fix fixed the problem from this issue. Hence closing this.

closed 2025-02-04 by urfeex