UniversalRobots/Universal_Robots_ROS2_Driver #1252: Teach pendant error message triggers protective stop

URJala, 2025-01-24:
### Affected ROS2 Driver version(s)

3.0.2

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS2 Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

Real robot, UR E-series robot, URSim in docker

### Robot SW / URSim version(s)

5.20

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Sometimes when getting the teach pendant error message mentioned in #1251 , a protective stop is triggered on the robot.

## Issue details
Sometimes the error message reported by the external control script is too long for the controller to handle. This triggers a protective stop, as it makes the servo_thread too slow. See the images.

[image]
[image]
[image]

## Steps to Reproduce
Trigger the error message as explained in #1251 until the protective stop is triggered. This can take some tries, as it depends on how the numbers in the message are rounded.

## Expected Behavior
The error message shouldnt trigger a protective stop.

## Actual Behavior
The error message slows down a real time thread and triggers a protective stop.

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-01-28:
So, the protective stop happens only, when the string is too long?

urfeex, 2025-01-29:
https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/256 could potentially help in mitigating this.

urfeex, 2025-01-31:
@URJala Could you please test whether https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/256 mitigates this?

urfeex, 2025-02-24:
Since https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/256 is merged, I will close this.

closed 2025-02-24 by urfeex