UniversalRobots/Universal_Robots_ROS2_Driver #1258: UR5 jerks after starting JointTrajectory Controller due to extremly high commands

taDachs, 2025-01-29:
### Affected ROS2 Driver version(s)

2.5.1

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

-

### How is the ROS driver used.

Headless without using the teach pendant

### Issue details

# Summary
We have a clearpath husky with a Ur5 mounted on top. When starting the joint trajectory controller and when sending a trajectory, the arm jerks violently and goes into a protective stop. The log contains extremely large joint commands:

[image]

## Steps to Reproduce
Start controller manager with a joint trajectory controller for the UR. Upon switching the controller on, the arm jerks. When continuing the program and sending a new trajectory, the arm again jerks and goes into a protective stop.

## Expected Behavior
The arm not jerking and executing the trajectory.

## Actual Behavior
The arm jerking, not executing the trajectory and going into a p-stop

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-01-29:
When saying "joint trajectory controller" do you mean "joint trajectory controller" or "scaled joint trajectory controller"?

How did you start the driver? Do you run it in one control node with the platform? That control frequency is too low, I experienced the same thing on exactly that robot a while back ;)

I am a bit puzzled though about the output. Those target numbers don't seam reasonable.

urfeex, 2025-01-29:
This issue was traced back to a too low frequency (50 Hz) leading to a lot of extrapolations. With this the robot was vibrating until it protective-stopped or hit the limit check.

While choosing the correct control frequency of 125 Hz didn't show the problem, https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/254 should robustify the connection against such things.

As far as I can see the issue is fixed for the OP, therefore closing this. @taDachs feel free to comment / reopen if you disagree.