UniversalRobots/Universal_Robots_ROS2_Driver #1259: Dual Arm Setup - Variable 'speed_slider_mask' is currently controlled by another RTDE client

FiIipe, 2025-01-31:
### Affected ROS2 Driver version(s)

Jazzy

### Used ROS distribution.

Rolling

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with lowlatency kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

5.17.3

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

Hello!

# Summary

Unable to run two UR5E robots in simultaneous - Dual Arm Setup - with ROS2 Jazzy.
Error: `Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured.`

## Issue details

Recently my team acquired two UR5e robots and my goal is to be able to run them with a dual arm setup. The robots are connected to a computer and I'm using URCaps - External Control to be able to communicate with them. I'm able to run the robots independently without problems. When I try to run both robots at the same time, I get the following error:
`Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured.`

I've read all the issues that I could find related to this "speed_slider_mask", - the Profinet and Ethernet/IP are disabled - but so far I'm out of luck. Also, the issue only shows up with real hardware, if I use mock hardware it works without problems.

*Detailed description help us understand the problem. Code are welcome!*

Launch file for robot 1 (walle_control.launch.py): 
```

# Copyright (c) 2021 PickNik, Inc.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
#    * Redistributions of source code must retain the above copyright
#      notice, this list of conditions and the following disclaimer.
#
#    * Redistributions in binary form must reproduce the above copyright
#      notice, this list of conditions and the following disclaimer in the
#      documentation and/or other materials provided with the distribution.
#
#    * Neither the name of the {copyright_holder} nor the names of its
#      contributors may be used to endorse or promote products derived from
#      this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
# ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
# LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
[... 1124 lines cut ...]
[ros2_control_node-1] #15   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in clone3 [0x7f35f7cdac3b]
[ros2_control_node-1] #14   Source "./nptl/pthread_create.c", line 447, in start_thread [0x7f35f7c4da93]
[ros2_control_node-1] #13   Source "../../../../../src/libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x7f35f7edddb3]
[ros2_control_node-1] #12   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7f35f81d9dc6, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[ros2_control_node-1] #11   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7f35f81c8c29, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[ros2_control_node-1] #10   Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7f35f81c84fa, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
[ros2_control_node-1] #9    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x7f35f842b614, in 
[ros2_control_node-1] #8    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x7f35f83bb43a, in controller_manager::ControllerManager::robot_description_callback(std_msgs::msg::String_<std::allocator<void> > const&)
[ros2_control_node-1] #7    Object "/opt/ros/jazzy/lib/libcontroller_manager.so", at 0x7f35f8390bc6, in 
[ros2_control_node-1] #6    Source "../../../../src/libstdc++-v3/libsupc++/eh_throw.cc", line 98, in __cxa_throw [0x7f35f7eac390]
[ros2_control_node-1] #5    Source "../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc", line 58, in terminate [0x7f35f7e96a54]
[ros2_control_node-1] #4    Source "../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc", line 48, in __terminate [0x7f35f7eac0d9]
[ros2_control_node-1] #3    Source "../../../../src/libstdc++-v3/libsupc++/vterminate.cc", line 95, in __verbose_terminate_handler [0x7f35f7e96ff4]
[ros2_control_node-1] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f35f7bd98fe]
[ros2_control_node-1] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f35f7bf626d]
[ros2_control_node-1] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-1]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-1]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f35f7c4fb1c]
[ros2_control_node-1] Aborted (Signal sent by tkill() 133778 1000)
[ERROR] [ros2_control_node-1]: process has died [pid 133778, exit code -6, cmd '/opt/ros/jazzy/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/eve --params-file /opt/ros/jazzy/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /tmp/launch_params_pqeszisk'].
```

### Accept Public visibility

- [x] I agree to make this context public

ayush5ingh, 2025-01-31:
I'm also facing same issue.

urfeex, 2025-02-03:
@FiIipe Your launch files look like you have a `robot_ip` argument, but you pass a hardcoded `10.0.0.10` to `ur_control.launch.py`.

@ayush5ingh unless you are using the exact same launchfile as the OP, I would suggest creating your own issue with all necessary details.

urfeex, 2025-03-04:
Closing this since a solution was proposed 1 month ago. Feel free to comment / reopen if the issue still exists.

closed 2025-03-04 by urfeex