UniversalRobots/Universal_Robots_ROS2_Driver #1267: URSim persistent settings

razr, 2025-02-04:
I'm not sure whether I can address it here, please advise.
I follow a description from https://hub.docker.com/r/universalrobots/ursim_e-series to keep my URSim settings persistent. 
I think there are some bits and pieces missing in the description. It looks like I need to 
- mount `default.installation` file `-v $HOME/ursim/default.installation`:/ursim/programs/default.installation` to keep installation settings
- mount `-v $HOME/ursim/.polyscope`:/ursim/.polyscope` to keep polyscope setting e.g. `Remote Control`
- mount  `-v $HOME/ursim/.urcontrol`:/ursim/.urcontrol` to keep  other settings e.g. a robot arm position.

1. Is it a complete list or I need to mount something else
2. I have tried to set `Network` to `Static Address` and it does not survive the reset
3. To modify and save an arm position does not work either for me

urfeex, 2025-02-04:
Hi, thank you for reaching out. An example of what you need to do for persistent storage can be found in [`start_ursim.sh`](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/88d31b9a0105035a1affd247b5f2e5f806b22296/scripts/start_ursim.sh#L208-L209)

1. We usually mount `/urcaps` and `/ursim/programs`.
2. The network address comes from the Docker container and cannot be set inside PolyScope
3. Saving an arm position isn't possible as far as I know.

@urrsk would you like to add anything to this?

urrsk, 2025-02-04:
@razr thanks for asking
The urp program file will, on load time, load trigger PolyScope to load the depended installation file. Mening that it will load the installation you last used when saving the program.

I normally start my ursim docker up with ```docker run --rm -it -v "${HOME}/programs:/ursim/programs" universalrobots/ursim_e-series``` and connect to it with the IP docker have assigned to it. You can give docker directions to use a static ip. See docker docs and script @urfeex pointed at as well.

In case it is still an issues, You are welcome to provide an example urp and installation files for us to reproduce your issue

razr, 2025-02-04:
Thanks for the explanations @urrsk @urfeex , let me look into it. Just for my understanding: the [`start_ursim.sh`](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/scripts/start_ursim.sh) script you have mentioned is not the same used in docker? I'm using the latest `universalrobots/ursim_e-serie` docker image 

```bash
~$ docker inspect universalrobots/ursim_e-series | grep -i version
        "DockerVersion": "20.10.17",
                "VERSION=5.20.0",
                "build_version": "URSim Version: 5.20.0",
```

```bash
# cat /ursim/start-ursim.sh
#!/bin/bash
export LD_LIBRARY_PATH=/opt/urtool-3.0/lib
URSIM_ROOT=$(dirname $(readlink -f $0))
pushd $URSIM_ROOT &>/dev/null
CLASSPATH=$(echo $URSIM_ROOT/lib/*.jar | tr ' ' ':')

./stopurcontrol.sh
./starturcontrol.sh
....
```

urfeex, 2025-02-05:
The `start_ursim.sh` script that I linked is a wrapper script around the docker image. This has nothing to do with the `/ursim/start-ursim.sh` script inside the container.

urfeex, 2025-02-06:
I'm going to close this, as I don't see any action to take here.

closed 2025-02-06 by urfeex

razr, 2025-02-06:
sorry, I still have some questions.

1. Why you do not mount `ursim/.polyscope/`? This is where PolyScope stores it's settings e.g. `remotecontrol.properties` etc.
2. I have tried to use `/urcaps` and copy `com.fzi.externalcontrol-1.0.5.jar` to it, however if I start ursim then `ExternalControl` appears in the list of inactive `URCaps`. It becomes active If I map it under `/ursim/.urcaps/`.
3. Is there a repo where the official Dockerfile for the ursim is located?

urfeex, 2025-02-07:
> sorry, I still have some questions.
> 
> 1. Why you do not mount `ursim/.polyscope/`? This is where PolyScope stores it's settings e.g. `remotecontrol.properties` etc.

As stated above, we don't really see any benefits from doing that. Remote control doesn't have any effect with the simulated robot, the network settings don't apply for the docker version.

> 2. I have tried to use `/urcaps` and copy `com.fzi.externalcontrol-1.0.5.jar` to it, however if I start ursim then `ExternalControl` appears in the list of inactive `URCaps`. It becomes active If I map it under `/ursim/.urcaps/`.

Please see the start script I linked earlier. That already injects that specific URCap. Is there a particular reason why you want to replicate this? Btw, the process of that file is further explained [here](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_client_library/doc/setup/ursim_docker.html#ursim-docker).

> 3. Is there a repo where the official Dockerfile for the ursim is located?

No, that is managed internally.

razr, 2025-02-07:
Thanks for your response and the link. I don't have ROS 2 installed on my host machine, which is why I didn't use your script.