UniversalRobots/Universal_Robots_ROS2_Driver #1272: gpio/analog_output_domain_cmd not existing

francescosemeraro, 2025-02-11:
### Affected ROS Driver version(s)

humble

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.13 (it does not matter)

### How is the ROS driver used.

Others

### Issue details

# Summary
I have downloaded again the ROS 2 Humble binary. ros2 launch ur_robot_driver ur_control.launch.py does not work anymore

## Issue details
I wanted to test an application of mine that I ran last July. Before coming back at it I reinstalled the UR package from binary.
When I simply try to run the launch file to spawn the fake hardware robot, I get this error, which is preventing me then from controlling the robot through the ur_moveit_config launch file:

[INFO] [launch]: All log files can be found below /home/francesco/.ros/log/2025-02-10-20-26-12-052574-francesco-ThinkPad-P53-8077
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [8079]
[INFO] [urscript_interface-2]: process started with pid [8081]
[INFO] [robot_state_publisher-3]: process started with pid [8083]
[INFO] [spawner-4]: process started with pid [8085]
[INFO] [spawner-5]: process started with pid [8087]
[ros2_control_node-1] [WARN] [1739219172.466669308] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1739219172.467323242] [resource_manager]: Loading hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1739219172.468268661] [resource_manager]: Initialize hardware 'ur5e' 
[ros2_control_node-1] [WARN] [1739219172.468325017] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-1] [WARN] [1739219172.468495943] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-1] <state_interface name="velocity"> 
[ros2_control_node-1]   <param name="initial_value">0.0</param> 
[ros2_control_node-1] </state_interface>
[ros2_control_node-1] [INFO] [1739219172.468517023] [resource_manager]: Successful initialization of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1739219172.468965677] [resource_manager]: 'configure' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1739219172.468991737] [resource_manager]: Successful 'configure' of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1739219172.469051931] [resource_manager]: 'activate' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1739219172.469065127] [resource_manager]: Successful 'activate' of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1739219172.473846078] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1739219172.473875191] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-1] [WARN] [1739219172.473981532] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[robot_state_publisher-3] [INFO] [1739219172.495134610] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1739219172.495217374] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1739219172.495227812] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-3] [INFO] [1739219172.495234281] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1739219172.495239748] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1739219172.495245399] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-3] [INFO] [1739219172.495250813] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1739219172.495256384] [robot_state_publisher]: got segment tool0
[robot_state_publisher-3] [INFO] [1739219172.495262100] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-3] [INFO] [1739219172.495267655] [robot_state_publisher]: got segment world
[... 77 lines cut ...]
[ros2_control_node-1] [INFO] [1739219172.891370525] [controller_manager]: Configuring controller 'forward_position_controller'
[ros2_control_node-1] [INFO] [1739219172.891683666] [forward_position_controller]: configure successful
[ros2_control_node-1] [INFO] [1739219172.893490182] [controller_manager]: Loading controller 'force_mode_controller'
[spawner-5] [INFO] [1739219172.899193220] [spawner_scaled_joint_trajectory_controller]: Loaded force_mode_controller
[ros2_control_node-1] [INFO] [1739219172.899529235] [controller_manager]: Configuring controller 'force_mode_controller'
[ros2_control_node-1] [INFO] [1739219172.907682294] [controller_manager]: Loading controller 'passthrough_trajectory_controller'
[spawner-5] [INFO] [1739219172.913047431] [spawner_scaled_joint_trajectory_controller]: Loaded passthrough_trajectory_controller
[ros2_control_node-1] [INFO] [1739219172.913282661] [controller_manager]: Configuring controller 'passthrough_trajectory_controller'
[ros2_control_node-1] [INFO] [1739219172.918391798] [controller_manager]: Loading controller 'freedrive_mode_controller'
[spawner-5] [INFO] [1739219172.925124419] [spawner_scaled_joint_trajectory_controller]: Loaded freedrive_mode_controller
[ros2_control_node-1] [INFO] [1739219172.925342263] [controller_manager]: Configuring controller 'freedrive_mode_controller'
[ERROR] [spawner-4]: process has died [pid 8085, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner --controller-manager /controller_manager --controller-manager-timeout 10 joint_state_broadcaster io_and_status_controller speed_scaling_state_broadcaster force_torque_sensor_broadcaster tcp_pose_broadcaster ur_configuration_controller joint_trajectory_controller --ros-args'].
[INFO] [spawner-5]: process has finished cleanly [pid 8087]

I have run into this issue in the past, but I can't seem to remember how I solved it. Can you please help me?

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-02-11:
Did you update all of your ROS packages? That particular missing command interfaces was [added to `ur_description` back in October](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/commit/32f618a37dfb9d07ce760e1c380a43f1c1dc9c8b).

I've just tested installing things from the binary packages. As we have just missed the merge window before the last sync, the currently released version for the driver is 2.5.1. The most notable thing missing is 588aa30. There will be error output from the pose broadcaster about `nan` values. You can silence them by stopping the `tcp_pose_broadcaster`, e.g. by calling 
```
ros2 service call /controller_manager/switch_controller controller_manager_msgs/srv/SwitchController "{deactivate_controllers: ['tcp_pose_broadcaster']}"
```

Apart from that, using fake hardware should work as normal.

urfeex, 2025-03-05:
Since the last sync this shouldn't pop up on humble anymore.

closed 2025-03-05 by urfeex