UniversalRobots/Universal_Robots_ROS2_Driver #1277: failed to use forward_position_controller control robot

migra1315, 2025-02-17:
### Affected ROS2 Driver version(s)

ubuntu22.04 + Ros humble +moveit -humble

### Used ROS distribution.

Humble

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

1.0.5

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I tried to use forward_position_controller to control robot, also have activated this controller, but it's no work

## Issue details
- The tech pendant perfrom text msg like "Velocity 3141.6005 required to reach the received target [xxxxxx] within 0.008 seconds is exceeding the joint velocity limits" 
- Besides, the content in  [xxxxxx] is not always same with the msg I've send to robot, like  "Velocity [A HUGE NUMBER] required to reach the received target [2.0,A HUGE NUMBER,A HUGE NUMBER,A HUGE NUMBER,xxx,xxx] within 0.008 seconds is exceeding the joint velocity limits" (in my works, I only change the 2nd,3rd,4th joint params)
- 
## Steps to Reproduce
```
# step 1 
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3 robot_ip:=192.168.157.10 launch_rviz:=false
# step 2 
ros2 control switch_controllers --deactivate scaled_joint_trajectory_controller --activate forward_position_controller
# step 3
# publish msg to  forward_position_controller
class JointPositionPublisher(Node):
    def __init__(self):
        super().__init__('joint_position_publisher')
        # 创建一个 QoS 配置文件，设置为 Transient Local
        qos_profile = QoSProfile(
            depth=10,  # 消息队列深度
            durability=DurabilityPolicy.TRANSIENT_LOCAL  # 设置为 Transient Local
        )
        self.publisher = self.create_publisher(Float64MultiArray, '/right/forward_position_controller/commands', qos_profile)
        self.timer = self.create_timer(1/200, self.publish_joint_positions)
        self.joint_positions = [2.8662843704223633,
            -2.4293726126300257,
            -2.2251551787005823,
            1.349448561668396,
            -1.015162769948141,
            1.9788888692855835
        ]
        self.step = 0.0005
        self.total_steps = 5000
        self.current_step = 0

    def publish_joint_positions(self):
        if self.current_step < self.total_steps:
            # Update the position of the second joint (elbow_joint)
            self.joint_positions[1] += self.step
            self.joint_positions[2] += self.step
            self.joint_positions[3] -= self.step
            # Create and publish the message
            msg = Float64MultiArray()
            msg.data = self.joint_positions
            self.publisher.publish(msg)
            self.get_logger().info(f'Published joint positions: {msg.data}')
            self.current_step += 1
        else:
            self.get_logger().info('Completed all steps.')
            self.timer.cancel()
```

## Expected Behavior
I'm sure the forward_position_controller has been activate...
Or may be the joint params is not suitable with real joints name list?

## Actual Behavior
*What did you observe? If possible please attach relevant information.*

# Workaround Suggestion
*If a workaround has been found, you are welcome to share it.*

### Relevant log output

```shell

```

### Accept Public visibility

- [x] I agree to make this context public

migra1315, 2025-02-17:
I want to make a addtion comments. I also tried to use moveit servo to control robot, which is also used forward_position_controller, and it's work well. I tried to copy the forward_position_controller/commands data recorded, but it's not work when I tried to publish same data to controller

urfeex, 2025-02-17:
Thank you for posting this. It seems that under some circumstances the limit check check can interpret the target buffer as very large numbers. I've been having trouble to reproduce this.

Does that happen for you in every script invocation? What is your robot's pose when you start that script? Obviously, if you command a too large step, the limit check will stop if from executing, but that shouldn't result in such big numbers as I am assuming, you don't command a step size of 25 rad ;-)

urfeex, 2025-02-17:
@migra1315 One more thing: When that happens, what does the "Variables" tab / Variables view show for `cmd_servo_q`? Does it also show the large numbers?

urfeex, 2025-02-17:
@migra1315 could you maybe also try out a branch that should give better feedback about the limit check: https://github.com/urfeex/Universal_Robots_Client_Library/tree/improve_limit_check

You would have to build the client library and driver from source:

* Remove the client library first (`sudo apt remove ros-humble-ur-client-library` -- this will remove the driver, as well)
* Build the client library and driver:
   ```
   export COLCON_WS=~/workspace/ros_ur_driver
   mkdir -p $COLCON_WS/src
   cd $COLCON_WS
   git clone -b humble https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/ur_robot_driver
   git clone -b improve_limit_check https://github.com/urfeex/Universal_Robots_Client_Library.git src/ur_client_library
   rosdep update
   rosdep install --ignore-src --from-paths src -y
   colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
   source install/setup.bash
   ```

Then, start the driver using:

```
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3 robot_ip:=192.168.157.10 launch_rviz:=false \
initial_joint_controller:=forward_position_controller
```

and run this script (I turned your snipped into a fully working minimal example):
```python
#!/usr/bin/env python3

import rclpy
from rclpy.node import Node
from rclpy.qos import QoSProfile
from rclpy.qos import DurabilityPolicy
from std_msgs.msg import Float64MultiArray

class JointPositionPublisher(Node):
    def __init__(self):
        super().__init__("joint_position_publisher")
        qos_profile = QoSProfile(
            depth=10,
            durability=DurabilityPolicy.TRANSIENT_LOCAL,
        )
        self.publisher = self.create_publisher(
            Float64MultiArray,
            "/forward_position_controller/commands",
            qos_profile,
        )
        self.timer = self.create_timer(1 / 200, self.publish_joint_positions)
        self.joint_positions = [
            2.8662843704223633,
            -2.4293726126300257,
            -2.2251551787005823,
            1.349448561668396,
            -1.015162769948141,
            1.9788888692855835,
        ]
        self.step = 0.0005
        self.total_steps = 5000
        self.current_step = 0

    def publish_joint_positions(self):
        if self.current_step < self.total_steps:
            # Update the position of the second joint (elbow_joint)
            self.joint_positions[1] += self.step
            self.joint_positions[2] += self.step
            self.joint_positions[3] -= self.step
            # Create and publish the message
            msg = Float64MultiArray()
            msg.data = self.joint_positions
            self.publisher.publish(msg)
            self.get_logger().info(f"Published joint positions: {msg.data}")
            self.current_step += 1
        else:
            self.get_logger().info("Completed all steps.")
            self.timer.cancel()

def main(args=None):
    rclpy.init(args=args)
    joint_position_publisher = JointPositionPublisher()
    rclpy.spin(joint_position_publisher)
    rclpy.logging.get_logger("testing").info("Shutting down...")
    joint_position_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()
```

migra1315, 2025-02-17:
thx for apply! May be I find the reason why msg "Velocity 3141.6005 required to reach the received target [xxxxxx] within 0.008 seconds is exceeding the joint velocity limits" occurs.
I set robot as 
```
name:
- shoulder_lift_joint
- elbow_joint
- wrist_1_joint
- wrist_2_joint
- wrist_3_joint
- shoulder_pan_joint
position:
- -1.5745227972613733
- 1.5622177124023438
- -0.005445782338277638
- 0.4171585142612457
- 0.0032716935966163874
- 0.0027926990296691656
```
and then pub msg to controller
```
ros2 topic pub --rate 1 /forward_position_controller/commands std_msgs/msg/Float64MultiArray "data: 
- 0
- -1.57
- 1.57
- 0.0
- 0.5
- 0.0"
```

the robot should perform as moving  immediately when get the first msgs right? But the robot didn't move util gets the 2nd or 3nd msgs.

Also, I have reorganized the initial code I provided, I think it's also move as the 10th or more further step, maybe it's the reason why teach pandent perform "velocity limit".(only one time, when I pub in 1Hz, the robot move as expected)

so sorry for I didn't understand the situation before asking a question！ But, the new problem still seems like abnormal...

> [@migra1315](https://github.com/migra1315) One more thing: When that happens, what does the "Variables" tab / Variables view show for `cmd_servo_q`? Does it also show the large numbers?

no, it's wired... Most of the time, it is a normal value, usually the current joint posture of the robot. But I encountered a special situation once as picture show

[image]

migra1315, 2025-02-17:
I found a strange solution, as I mentioned in my previous comment, that works fine when the timer cycle is set to 1 Hz...
so I try replace 1/125 to 0.008, and this is works...
I have not change anything but the timer init method...
```
# change this 
self.timer = self.create_timer(1/125, self.publish_joint_positions)
# into 
self.timer = self.create_timer(0.008, self.publish_joint_positions)
```
I will continue to test this plan and release the latest progress

migra1315, 2025-02-19:
I have made some attempts again, and now this strategy can be used normally.
The main codes are as follows:
```
class JointPositionPublisher(Node):
    def __init__(self):
        super().__init__('joint_position_publisher')
        self.positions = {}

        # 创建一个 QoS 配置文件，设置为 Transient Local
        qos_profile = QoSProfile(
            depth=10,  # 消息队列深度
            durability=DurabilityPolicy.TRANSIENT_LOCAL  # 设置为 Transient Local
        )
        self.publisher = self.create_publisher(Float64MultiArray, '/forward_position_controller/commands', qos_profile)

        time.sleep(1)
        self.get_logger().info(f'wait 1 sec ...')

        time.sleep(1)
        self.get_logger().info(f'wait 2 sec ...')

        time.sleep(1)
        self.get_logger().info(f'wait 3 sec ...')

        self.timer = self.create_timer(0.008, self.publish_joint_positions)

        self.joint_positions = [0.0, -1.57, 2.57, -2.0, -2.0, 0.0]

        self.step = 0.03
        self.total_steps = 100
        self.current_step = 0

    def publish_joint_positions(self):
        if self.current_step < self.total_steps:
            # Update the position of the second joint (elbow_joint)
            self.joint_positions[2] += (self.step/2)
            self.joint_positions[3] += self.step
            self.joint_positions[4] += self.step
            # Create and publish the message
            msg = Float64MultiArray()
            msg.data = self.joint_positions

            self.publisher.publish(msg)
            self.current_step += 1
            self.get_logger().info(f'Published joint positions: {msg.data}')

        else:
            self.get_logger().info('Completed all steps.')
            self.timer.cancel()
```
The most critical part is to wait for a short period of time after creating the publisher to ensure successful connection and avoid the robotic arm receiving a large displacement (as the first few frames of data may be lost).

my test results are relatively stable ... Thank you for your assistance

urfeex, 2025-02-19:
OK, so Waiting for the publisher to establish the connection helps you getting reliable behavior? That would mean that the limit check actually did what it is supposed to do, as it was preventing the robot to move a large displacement.

However, that would't explain the >3000 rad/sec velocity you reported initially.

migra1315, 2025-02-24:
It should works like this: 

The PC sends a control message, and the robot can't receive that message until the connection is established. The PC expects the robot to gradually run from 0 to 2 in step 0.1, but the robot receives instructions to move directly to 2 because none of the previous instructions were received. I guess that's why the bot reports a huge speed number error.

urfeex, 2025-02-24:
And that's exactly why we have the limit check in place. As otherwise the robot would move to the target position as fast as possible.

In that case I think we can close this issue. Feel free to comment / reopen if you disagree. Thank you for coming back to this.

closed 2025-02-24 by urfeex