UniversalRobots/Universal_Robots_ROS2_Driver #1279: Universal_Robots_ROS2_Driver.humble.repos has a flat directory structure

razr, 2025-02-19:
Currently  [Universal_Robots_ROS2_Driver.humble.repos](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/Universal_Robots_ROS2_Driver.humble.repos) has a flat structure. It might be better to have a structure similar to [ros2.repos](https://github.com/ros2/ros2/blob/humble/ros2.repos)

```bash
src$ tree -L 2
.
├── ros-controls
│   ├── control_msgs
│   ├── control_toolbox
│   ├── kinematics_interface
│   ├── realtime_tools
│   ├── ros2_control
│   └── ros2_controllers
├── ros-industrial
│   └── ur_msgs
└── UniversalRobots
    ├── Universal_Robots_Client_Library
    └── Universal_Robots_ROS2_Description
```

urfeex, 2025-02-20:
The repos file is mainly meant for CI testing, so I don't necessarily see the benefits of structuring this.

razr, 2025-02-20:
np, I just read in the [installation](https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_robot_driver/ur_robot_driver/doc/installation/installation.html#build-from-source) it can be used for the source build, and I found it inconvenient to use in this form.

urfeex, 2025-03-05:
I'm closing this as a wontfix if you don't mind.

closed 2025-03-05 by urfeex