UniversalRobots/Universal_Robots_ROS2_Driver #1290: Connect o3de with ur

TheMassimo, 2025-03-05:
### Affected ROS2 Driver version(s)

latest

### Used ROS distribution.

Other

### Which combination of platform is the ROS driver running on.

Ubuntu Linux with standard kernel

### How is the UR ROS2 Driver installed.

From binary packets

### Which robot platform is the driver connected to.

URSim in linux, Real robot, URSim in a virtual machine, URSim in docker, UR E-series robot, UR CB3 robot

### Robot SW / URSim version(s)

latest

### How is the ROS driver used.

Others

### Issue details

# Summary
I want to use the default example of ur robot in o3de extra and move it with rviz

## Issue details
In o3de-extras there is a example that include universal robot: https://github.com/o3de/o3de-extras/tree/development/Templates/Ros2RoboticManipulationTemplate. There is also a launch file but is only for panda robot. So I try to create a launch file following that one. But it does not work and my launch wait for /robot_description topic. Also there are some warning at the start that i'don't undestand why

## Steps to Reproduce
download o3de and o3de-extras (latest version 2409.2). Download ur_drivere and ur_movit (latest version). Now run o3de project and run the launch file (ros2 launch my_test.launch.py ur_type:=ur10)

## Expected Behavior
The flow should be the same as the provided in panda example

## Actual Behavior
it doesn't work

##My launch file
import os
import yaml

from pathlib import Path

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, RegisterEventHandler
from launch.conditions import IfCondition
from launch.event_handlers import OnProcessExit
from launch.substitutions import (
    LaunchConfiguration,
    PathJoinSubstitution,
)

from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

from moveit_configs_utils import MoveItConfigsBuilder

from ament_index_python.packages import get_package_share_directory

from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

[... 180 lines cut ...]
                    robot_state_publisher,
                    rgbd_pc,
                ],
            )
        ),
    )

    return ld

### Relevant log output

```shell
[INFO] [launch]: All log files can be found below /home/robx/.ros/log/2025-03-05-19-35-14-534067-robx-MS-7D67-316208
[INFO] [launch]: Default logging verbosity is set to INFO
WARNING:root:Cannot infer URDF from `/opt/ros/jazzy/share/ur_moveit_config`. -- using config/ur.urdf
WARNING:root:Cannot infer SRDF from `/opt/ros/jazzy/share/ur_moveit_config`. -- using config/ur.srdf
WARNING:root:"File /opt/ros/jazzy/share/ur_moveit_config/config/ur.urdf doesn't exist"
WARNING:root:The robot description will be loaded from /robot_description topic 
[INFO] [wait_for_robot_description-1]: process started with pid [316211]
[wait_for_robot_description-1] [INFO] [1741199714.811761162] [wait_for_robot_description]: Waiting for message on /robot_description.
```

### Accept Public visibility

- [x] I agree to make this context public

urfeex, 2025-03-06:
Hi, thank you for reaching out. We appreciate that you try to use other simulators with a UR robot. However, at the moment we only support Gazebo as a simulator.

However, your questions seems more to be related to your custom launch file than the the simulator. Please understand that we also cannot help individual problems when creating custom integrations. Looking at your output, however, it seems that the move_group / the helper node is waiting for the robot description being published. In the driver launch that comes from the robot_state publisher launched in [rsp.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/launch/ur_rsp.launch.py).

I don't know how o3de integrates with a URDF / robot_description, so this could be wrong advice. You can also make the move_group load the description. For that, it is easiest to create a custom moveit support package using the moveit setup assistant. The default launchfiles generated from that will load the description. See out [custom workcell tutorial](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_tutorials/my_robot_cell/doc/build_moveit_config.html) on details for that.

jhanca-robotecai, 2025-03-18:
The main difference between O3DE and Gazebo is that O3DE does not publish robot descriptions on `/robot_description` topic. 

The `ur_moveit.launch.py` script worked fine on ROS 2 Humble (`humble` branch) without getting any information on the topic, but it was refactored on the `main` branch and this information is required. I do not know the history behind this change, but we do not have the resources in the O3DE community to maintain the _launchers_ of other robots, as we focus on developing the engine itself. The information about UR robots will be removed from the O3DE documentation.