UniversalRobots/Universal_Robots_ROS2_Driver #1293: Colcon build issue

hcrane2093, 2025-03-11:
Hi

I am trying to build these packages on ROS2 humble for my dissertation at university. 

Despite being able to do this with other robot packages such as the Franka Emika Robot packages, the Humble packages using this repo seem to fail every time. 

The main errors I keep getting back are like this and I do not understand them. It looks like Chat GPT 03-mini-high does not know what to do either...

Please let me know if you could help.

Thanks 

hc@hc-OMEN-Transcend-Gaming-Laptop-14-fb0xxx:~/ur_ws$ colcon build --symlink-install
Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [0.61s]                                                               
Finished <<< ur_dashboard_msgs [7.75s]                     
Starting >>> ur_controllers
--- stderr: ur_controllers                              
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp: In member function ‘virtual controller_interface::CallbackReturn ur_controllers::URConfigurationController::on_activate(const rclcpp_lifecycle::State&)’:
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp:97:30: error: cannot convert ‘ur_controllers::URConfigurationController::on_activate(const rclcpp_lifecycle::State&)::<lambda(std::shared_ptr<ur_controllers::VersionInformation>)>’ to ‘const std::shared_ptr<ur_controllers::VersionInformation>&’
   97 |   robot_software_version_.set([this](const std::shared_ptr<VersionInformation> ptr) {
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   98 |     ptr->major = state_interfaces_[StateInterfaces::ROBOT_VERSION_MAJOR].get_value();
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   99 |     ptr->minor = state_interfaces_[StateInterfaces::ROBOT_VERSION_MINOR].get_value();
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  100 |     ptr->build = state_interfaces_[StateInterfaces::ROBOT_VERSION_BUILD].get_value();
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  101 |     ptr->bugfix = state_interfaces_[StateInterfaces::ROBOT_VERSION_BUGFIX].get_value();
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  102 |   });
      |   ~~                          
In file included from /home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/include/ur_controllers/ur_configuration_controller.hpp:47,
                 from /home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp:41:
/opt/ros/humble/include/realtime_tools/realtime_tools/realtime_box.hpp:51:22: note:   initializing argument 1 of ‘void realtime_tools::RealtimeBox<T>::set(const T&) [with T = std::shared_ptr<ur_controllers::VersionInformation>]’
   51 |   void set(const T & value)
      |            ~~~~~~~~~~^~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp: In member function ‘bool ur_controllers::URConfigurationController::getRobotSoftwareVersion(ur_msgs::srv::GetRobotSoftwareVersion_Request_<std::allocator<void> >::SharedPtr, ur_msgs::srv::GetRobotSoftwareVersion_Response_<std::allocator<void> >::SharedPtr)’:
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/ur_configuration_controller.cpp:117:34: error: ‘class realtime_tools::RealtimeBox<std::shared_ptr<ur_controllers::VersionInformation> >’ has no member named ‘try_get’
  117 |   return robot_software_version_.try_get([resp](const std::shared_ptr<VersionInformation> ptr) {
      |                                  ^~~~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp: In member function ‘virtual controller_interface::CallbackReturn ur_controllers::PassthroughTrajectoryController::on_deactivate(const rclcpp_lifecycle::State&)’:
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:212:49: error: could not convert ‘(&((ur_controllers::PassthroughTrajectoryController*)this)->ur_controllers::PassthroughTrajectoryController::abort_command_interface_.std::optional<std::reference_wrapper<hardware_interface::LoanedCommandInterface> >::operator->()->std::reference_wrapper<hardware_interface::LoanedCommandInterface>::get())->hardware_interface::LoanedCommandInterface::set_value(1.0e+0)’ from ‘void’ to ‘bool’
  212 |   if (!abort_command_interface_->get().set_value(1.0)) {
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
      |                                                 |
      |                                                 void
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:212:49: error: in argument to unary !
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::PassthroughTrajectoryController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:256:68: error: void value not ignored as it ought to be
  256 |       write_success &= transfer_command_interface_->get().set_value(TRANSFER_STATE_WAITING_FOR_POINT);
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:265:77: error: void value not ignored as it ought to be
  265 |         write_success &= time_from_start_command_interface_->get().set_value(
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  266 |             duration_to_double(active_joint_traj_.points[current_index_].time_from_start));
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:273:64: error: void value not ignored as it ought to be
  273 |           write_success &= command_interfaces_[i * 3].set_value(
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  274 |               active_joint_traj_.points[current_index_].positions[joint_mapping->at(joint_names_internal->at(i))]);
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:277:70: error: void value not ignored as it ought to be
  277 |             write_success &= command_interfaces_[i * 3 + 1].set_value(
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  278 |                 active_joint_traj_.points[current_index_].velocities[joint_mapping->at(joint_names_internal->at(i))]);
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/passthrough_trajectory_controller.cpp:280:72: error: void value not ignored as it ought to be
  280 |               write_success &= command_interfaces_[i * 3 + 2].set_value(
[... 169 lines cut ...]
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/freedrive_mode_controller.cpp:214:75: error: void value not ignored as it ought to be
  214 |         write_success &= async_success_command_interface_->get().set_value(ASYNC_WAITING);
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/freedrive_mode_controller.cpp:221:65: error: void value not ignored as it ought to be
  221 |       write_success &= abort_command_interface_->get().set_value(1.0);
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/home/hc/ur_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/freedrive_mode_controller.cpp:223:73: error: void value not ignored as it ought to be
  223 |       write_success &= async_success_command_interface_->get().set_value(ASYNC_WAITING);
[... cut ...]
gmake[2]: *** [CMakeFiles/ur_controllers.dir/build.make:118: CMakeFiles/ur_controllers.dir/src/freedrive_mode_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:380: CMakeFiles/ur_controllers.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ur_controllers [6.20s, exited with code 2]
                                 
Summary: 2 packages finished [14.1s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  3 packages not processed
hc@hc-OMEN-Transcend-Gaming-Laptop-14-fb0xxx:~/ur_ws$ 
---- please help again

urfeex, 2025-03-12:
Thank you for reaching out. The output looks as if you are trying to build the `main` branch on Humble. When using ROS Humble, please use the `humble` branch.

urfeex, 2025-03-17:
Glad to hear, you got something working. Without any log output, we cannot really help you with the errors you seem to be facing.

Please understand, that I cannot give individual help regarding ROS usage. We do have an example to setup a custom workcell in our [Rolling documentation](https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/doc/ur_tutorials/my_robot_cell/doc/index.html). A humble version is pending, but the concepts should be the same. For the current status of the humble version please see https://github.com/UniversalRobots/Universal_Robots_ROS2_Tutorials/pull/11.

hcrane2093, 2025-03-17:
Thanks very much for that advice I will have a look at both of those options!!

H

closed 2025-03-24 by urfeex